TomWhitwell/RadioMusic #40: CV issues

mezzman, 2015-01-18:
I am building a bunch of RadioMusic modules (rev2 PCBs) and I am having odd issues with a few of them. I am seeing the Reset LED and bank indicator LEDs flash while a sample is playing. Sometimes this happens with CV into either Start or Tuner position inputs. Sometimes running audio out direct into an external mixer will cause this (ground loop?), but going into a VCA before the mixer fixes it.
Other times changing voltage going into Start actually triggers reset on voltage change, and I see the reset LED flash with the upper LEDs flashing the current bank (on top of the VU display)
And other times just a constant voltage into the tuner position input causes these to flash very rapidly and 
audio starts stuttering.

TomWhitwell, 2015-01-18:
I've started a [troubleshooting page in the wiki](https://github.com/TomWhitwell/RadioMusic/wiki/Troubleshooting-the-Radio-Music-module)  
And added a few things that might be relevant to you.  
Because you're the first - everything on that page is relevant to your questions! 

You can get stuttering 'between' stations - either from knob position or from CV input - I haven't yet found any way to avoid that 

I'm going to close this issue now, considering it a question answered, but be sure to re-open it if you need more help. 

I'll leave issues open once we identify a hardware or software issue that needs to be fixed

mezzman, 2015-01-18:
I ran the test sketch and I am seeing an identical issue across several boards.
What I am seeing is Channel CV and Time CV are noisy with nothing plugged into the jacks. LEDs 3&4 are flickering.
Channel CV ranges from 1 to 280 and Time CV is 1 to 400. Introducing CV into Channel jack makes both LEDs 3&4 change intensity, but only Channel CV changes in value. It still is noisy, as it jumps around 20-30 points. CV into Time jack has no effect on LED 3&4 intensity, but value does change and is also noisy.

TomWhitwell, 2015-01-18:
Hi, take a look at issue #13 re: values for R21 and R10 
https://github.com/TomWhitwell/RadioMusic/issues/13

Also, if you have Rev1 boards, there is a ground error: 
https://github.com/TomWhitwell/RadioMusic/wiki/PCB-Version-History

See the fix here regarding the legs on R21 and R10: 
https://github.com/TomWhitwell/RadioMusic/wiki/Building-the-Radio-Music-module

mezzman, 2015-01-18:
I am using rev2 boards and the following resistor values:
R10 47k 
R21 47k 
R9 22k 
R20 22k
I can try different values, would going lower be a good place to start?

TomWhitwell, 2015-01-18:
Those values work fine with my setup, so I don't think that's the problem.

mezzman, 2015-01-18:
Are you getting individual LED control for each CV input on the test?

TomWhitwell, 2015-01-18:
Hi,
I've updated the test sketch (there was a bug in the LED display, sorry) 
And made a demo video to show how it's supposed to work: 
https://github.com/TomWhitwell/RadioMusic/tree/master/Collateral/ModuleTest

mezzman, 2015-01-19:
Thank you. I downloaded the updated test sketch and now I am getting a curious issue. On a module that had no apparent noise problems at the CV inputs with the old test sketch now shows noise at the Time CV LED with nothing plugged in there, and goes away with 0v input via external source.
I am measuring -.1v between pin20/A6 and gnd.

TomWhitwell, 2015-01-19:
I sounds a lot like a hardware grounding issue; a bad connection somewhere. The ground path goes all the way through the Teensy itself, so it could be anywhere on the boards.

TomWhitwell, 2015-01-19:
On an unplugged module, put a patch lead into one of the CV inputs, and use a multimeter to check the resistance between tip and sleeve. It should be 47K

mezzman, 2015-01-19:
I am getting around 43k on both CV inputs.

monobass, 2015-01-19:
I'm seeing 47k on both Station and Start.

mezzman, 2015-01-19:
I removed the TL074 and connecting pin 8 on the socket to GND stops the noise. This pretty much rules out bad grounding. I am suspecting the opamp is the culprit, will try to find some locally today and swap it out.

mezzman, 2015-01-19:
I tried a different manufacturer of TL074 and also a TL084 with no improvement. back to square one. I just do not know why this is happening with all boards that I'm having this negative voltage offset which is causing the noise.

mezzman, 2015-01-21:
I am going to do a separate build with PCBs from a different source to test with, I will close this issue now.

monobass, 2015-01-29:
mezzman what value pots did you use?

mezzman, 2015-02-01:
50k linear Alps (from small bear) on both pots. Yes I had grayscale panel installed. I removed panel and there was no difference

Thonk-Steve, 2015-02-01:
Ah I wondered cos I ran into some cv drifting issues when using metal-shafted 100k pots, particularly on the Start pot... And even more so when the panel was off and they weren't grounded. Once I swapped back to 50k and put the panel back on the problems disappeared again.

mezzman, 2015-02-01:
That's interesting. The issues I have had were only on the CV inputs, the pots were rock solid.

Thonk-Steve, 2015-02-01:
Yeah they are rock solid now. I updated the bom to warn people off using 100k pending further testing.

mezzman, 2015-02-01:
I would also say stay away from Itead as a PCB fabricator, I used a PCB made by Seeed and it worked perfectly with no noise issues at the CV inputs. I have also had confirmation from others that Itead has had quality control issues.

Thonk-Steve, 2015-02-01:
Oh yeah iTead are definitely worth swerving! Their boards have been pretty poor for a couple of years now. Seeed protos on average much better from a manufacturing standpoint, the silkscreen on protos is pretty hit and miss. This issue should be closed then?

mezzman, 2015-02-01:
Yes, this issue has been resolved as far as I can tell. Progress has been made.

closed 2015-02-01 by mezzman