TomWhitwell/RadioMusic #44: Crossfade / Polyphonic mode

TomWhitwell, 2015-02-02:
To get smooth cuts when switching channels or starting at a different time, we need some kind of crossfade or zero crossing system   
At the moment (1.1) it works like this:  

1.  CV/Trigger received  (asking module to shift from audio A to audio B) 
1.  CV/Trigger noticed by the module polling the inputs 
1.  Start fadeOut Audio A
1.  Delay for crossfade time, until audio level is zero  
1.  Start to play Audio B, start to fadeIn  
1.  Delay for crossfade time, until audio level is full 
1.  Continue! 

This adds a bit of delay - a few milliseconds, controllable in settings.txt  

I think it would be improved - with some extra benefits - with a polyphonic mode, where multiple streams can play from the DAC output simultaneously  

This is how it might work: 

1.  Audio A is playing from output 1. Outputs 1, 2, 3, 4 are all mixed together to the physical mono output. 
1.  CV/Trigger received  (asking module to shift from audio A to audio B) 
1.  CV/Trigger noticed by the module polling the inputs 
1.  Simultaneously: 
    1.  Start playing Audio B from output 2
    1.  Start fading Audio B up (at 'Attack' speed) 
    1.  Start fading Audio A down (at 'Decay' speed) 
1.  No need to delay - fadeIn and fadeOut complete in the background  
1.  Cycle through the active outputs - number of possible outputs will be limited by the SD card read ability. According to the [Teensy Audio Shield Documentation](https://www.pjrc.com/store/teensy3_audio.html) you should be able to play 2-4 simultaneous wav files from a good quality card.  So we need some kind of simple voice allocation system. At first it might just be 'When playing A, start playing B, When playing B, start playing A', but we might as well write it to allow n simultaneous voices. Having 4 x simultaneous voices out of one mono output will be interesting / useful for drum programming and pads  

Need to think through:  

1.  Mixing levels to ensure don't suddenly get double loud every time you xfade 
1.  Retrigger logic - do we need a legato mode vs mono mode?

Thonk-Steve, 2015-02-02:
I think this is a great idea, getting a nice equal power curve on the crossfade would be the key.

Also would be great for a nice sounding oneshot/drum type trigger too.

mxmxmx, 2015-02-02:
"At first it might just be 'When playing A, start playing B, When playing B, start playing A'"

this is easy. one straightforward way would be to change your audio things to something like:

AudioPlaySdRaw           raw1, raw2;  
AudioPlaySdRaw           *raw[2] = {&raw1, &raw2};
AudioEffectFade            fade1; fade2;
AudioEffectFade            *fade[2] = { &fade1, &fade2};

and some mixer etc. then you could simple toggle the player object at each incoming trigger:

fade[_numVoice]->fadeIn(f_in_ms); 
wav[_numVoice]->playFrom(name, pos); 
_numVoice = ~_numVoice & 1u;
fade[_numVoice]->fadeOut(f_out_ms);

that's a fairly crude crossfade but it'll probably do.

it would be easy to extend that to more voices, maybe 4 or 5, in which case _numVoice might be a little ring buffer or counter. the thing i'm wondering about is would this work in practice? ie there's no way to set the file-end position or anything equivalent to 'sustain'. so when you have a file that's, say, 10 minutes long, you'd end up having all 4 files playing, and the fifth trigger then just kills the first voice ... might work for short samples, but otherwise it'll just be cacaphony, i imagine.