TomWhitwell/RadioMusic #46: Replacing pots with encoders

Thonk-Steve, 2015-02-02:
With thoughts of how Radio Music could be reprogrammed to be other things, would replacing the pots with rotary encoders require only software changes?

mxmxmx, 2015-02-02:
no.
encoders are wired differently.

Thonk-Steve, 2015-02-02:
I need to read up on encoders :)

Nantonos, 2015-02-02:
Hello Thonk-Steve,

Monday, February 2, 2015, 1:34:55 PM, you wrote:

> I need to read up on encoders :)

Encoders typically use two digital input pins, each one with a train
of pulses. From the phase response of the two, not only the rate of
turning but also the direction can be determined.

http://en.wikipedia.org/wiki/Rotary_encoder#Incremental_rotary_encoder

There is an Encoder library which can be used on teensy.
http://www.pjrc.com/teensy/td_libs_Encoder.html

But the PCB layout for an encoder and a pot are different. A pot is
typically connected to some constant voltage and to GND, with the
variable voltage on the wiper being read by an analog input.

-- 
Best regards,
 Nantonos                            mailto:nantonos@epona.net

mxmxmx, 2015-02-02:
short story short: it wouldn't involve very drastic changes, just rerouting a few traces. but then encoders are kind of pointless without some kind of display (?)

Thonk-Steve, 2015-02-02:
There's a push button and 4 leds up the top, with an encoder you could have 4 different modes quite simply

Thonk-Steve, 2015-02-02:
or obviously you could do a latching kind of thing with a pot... but I was never so keen on that.

TomWhitwell, 2015-02-03:
Going to close this in the interests of tidiness!

closed 2015-02-03 by TomWhitwell