TomWhitwell/RadioMusic #47: Separate Settings.txt for each bank

TomWhitwell, 2015-02-02:
This was suggested by Steven, I think it's probably a good idea so long as it's easy to ignore.  

I think you might get the module to write a settings.txt into each folder, plus the global one at the root (they might have different file names, but would have exactly the same variables) 

The first variable would be "Local" 

Local=0, the module ignores the local setttings.txt, and uses the global version   
Local=1, the module ignores the global settings and reads the local version in the folder  

As with the global version, if the file isn't found, it creates a new default one.

xavierbonfill, 2019-02-03:
I would be very interested on this feature! Can it work with the 2017 firmware or does it require recoding it?
Thanks in advance and keep up the great work!

Normalised, 2019-02-03:
It would require some development work.

xavierbonfill, 2019-02-03:
Thanks for the reply :) what would exactly need to be done?

Normalised, 2019-02-03:
First thing is to work out if you scan and store the settings at boot or if you do it on demand as the user changes bank. The trade off is memory usage (keeping all settings for each bank in memory) vs bank switch delay (reading a file from SD every time you switch bank).

Once you work that out, you just have to write the code :)

There may be some issues with changing settings on the fly as currently the code assumes that once they're set, they don't change. It's been some time since I looked at it, so I can't be 100% sure.

If you wan't to tackle it, i'll happily help out as much as I can.

PZHRHG, 2019-05-13:
very useful feature indeed .. 
best of the best would be to have looping defined file by file as most of the drum SW have , 
but may be is too much .
Can it be done with file naming convention ? crappy idea ?

dessertplanet, 2024-06-18:
I've had a crack at this! check it out https://github.com/TomWhitwell/RadioMusic/pull/192