TomWhitwell/RadioMusic #50: Hum when module is in a separate case from output mixer

TomWhitwell, 2015-02-07:
So, I've now been able to reproduce this after the reports today.   

I have two cases; one Doepfer LC9, one skiff with Tiptop Uzeus   

I have a Unify output mixer in the LC9.   

Radio Music in the LC9 with the audio going to the Unify = No problem   
Radio Music in the Skiff with audio going directly to speakers = No problem   
Radio Music in Skiff, with audio going to Unify in the LC9 = HUM   
  
The Hum stops as soon as I attach any other cable from the LC9 to the Radio Music (i.e. connect the output of an unused filter to the Reset or a CV input)   

Sometimes while experimenting, I need to cycle power in the skiff to get the module working properly.   

So:   

Workaround hypotheses:   
a) Have the Radio Music in the same case as your output mixer   
or:   
b) Plug something into one of the CV inputs   

Next: Further experimentation to get to the bottom of the problem

Nantonos, 2015-02-07:
[quote="Tombola"]
I have two cases; one Doepfer LC9, one skiff with Tiptop Uzeus 

The Hum stops as soon as I attach any other cable from the LC9 to the Radio Music (i.e. connect the output of an unused filter to the Reset or a CV input) [/quote]

The hum stops as soon as a cable shield is used to make a firm 0V connection so that ground noise from uzeus is diverted to the more solid ground of the LC9 without going via your module.

Although a cable shield works (and multiple cable shields work better) a substantial stranded cable going between the 0V of the two power supplies will give the best results.

TomWhitwell, 2015-02-08:
Connecting any module in the uZeus skiff to any other module in the main case kills the hum. 

Connecting the output of any single module in the skiff to the main mixer = hum (I mainly use it for sequencing, do don't do this very often!) 

Connecting ground in the main case to ground in the uZeus skiff = kills the hum

TomWhitwell, 2015-02-19:
Further experimentation on the rear PCB. Curiouser and curiouser. 

Looking again at the rear PCB, the ground path is quite convoluted, so I soldered a wire from the GND leg of one of the power diodes, to the DGND leg of one of the protection diodes. 

I then installed the module my separate case. 

Instead of hum, there was complete silence. Otherwise the module behaved as before - coming to life when a second cable was connected to the module. It worked completely as normal in the same module as the output mixer. 

I also noticed that the module comes to life when the tip of a patch cable touches the ring of a socket. It's as if the output is 'floating' wrongly. 

Continuing to experiment.

TomWhitwell, 2015-02-20:
Another test. I wondered if adding the second ground path would reduce noise. 

Not that I can measure: I calibrated the 440hz sine to 4.5v p-to-p on two modules, one with the fix, one without. 
The silent sample was showing 16-17mv RMS, including  a 10mv offset, in the modified module 
It showed 12mv including a 6mv offset in the unmodified module.

dinsync, 2015-02-25:
This may hopefully offer some insight, I also had the hum going from one case to another and was able to eliminate it using a simple non inverting voltage follower.

Paul

celestialtomes, 2015-05-25:
any progress on this? Haven't had time myself to troubleshoot

celestialtomes, 2015-05-27:
by any chance could you take a pic or show a schem of where you applied the non-inverting voltage follower?

I get the hum even when it is connected to modules with the same power source.

dinsync, 2015-05-28:
im not actually sure now in hindsight if this is making a difference as
I've noticed a slight hum. But anyway something like this

http://www.electronics-tutorials.ws/opamp/opamp17.gif?81223b

TomWhitwell, 2015-05-28:
I haven't done much work on this yet. 

I think the issue may be that the DAC output is floating  - there is no clear connection between the DAC and IC4D and the audio-out socket ground pin, which might be the only connection to ground in a second case.

However, I DON'T REALLY KNOW WHAT THAT MEANS, IF IT MEANS ANYTHING. 

I think that adding the 100k resistor path to ground parallel to C4 (see #70 ) is relevant to this, because it seemed to reduce the issue for me. 

My other initial thought was that the ground path on the rear PCB is rather torturous, and goes through the ferrite bead on the Teensy itself. But I've tried shorting this ground path with a fat wire, and it made no difference. 

Connecting a second patch cable between other modules in the two cases (or the case and the audio interface/amp) usually solves the problem. 

@dinsync This specific issue isn't usually a little slight hum when you turn the volume up, it's a massive hum with little or no audio signal audible.

dinsync, 2015-05-28:
yes i think the hum I'm hearing is something different, its there and can
be audible but perhaps not the same as this after all

djamsia, 2016-02-04:
I recently set up a Radio station.
there is no silence when no sounds, there is always a ground noise.
also get a loud hum from the reset input. when it used as an output.
I have the pcb rev2.1

elektronic2000, 2016-09-15:
@TomWhitwell Hello Tom i also have hum noises im interest in try soldering a cable as you mention in past post do you have any pic showing where to solder unseemly im not that professional to understand 100% the schematic will be very helpful with a pic.. thanks for your time, by the way i love this module ;)

NickShaffner, 2017-06-16:
Same here, still seeing this issue on my fresh v2.1 build.   

On the scope I'm seeing the correct signal on the teensy Adc but the noise/hum seems to be introduced between the adc pin and the audio out header pin on the bottom board.  

Has there been any movement on this?

BobJack, 2018-02-22:
the same question...

arnorhs, 2020-09-15:
same issue for me - constant hum, red thonk version

arnorhs, 2020-09-15:
i applied the 100k resistor hack from #70 and that seems to have gotten rid fo the hum for me

BobJack, 2020-09-15:
thanks for your reply!

I did the same, but after that the module began to turn on once every other
time

abramjanzie, 2020-10-31:
I am having this same issue. Fresh build, PCB v2.1, constant hum/noise. I've added the 100k resistor as per #70 but that does not address this issue. 

The LEDs on top seem to be reacting to the files that I have loaded on the SD card (I intentionally used a speech to make it as obvious as possible), but the only sound is a loud hum/noise that minimally reacts to CV input. 

All soldering joints are solid; all diodes, transistors, LEDs, and ICs are correctly oriented. 

This happens with both the Radio Music firmware and the test module firmware.