TomWhitwell/RadioMusic #51: Station pot/Tune CV in should only execute the station change to a different one

PawelGrudzien, 2015-02-12:
This is somewhat related to crossfade mode (issue #44).

Whenever the Station pot position changes (or external Tune control voltage changes), the module attempts to change the station, even if the CV difference is small enough that it is still within the range of the same station.

An example:

Even with only one sound file in the bank, turning the Station knob (or applying slow LFO to Tune CV input) causes stutter, as the module changes the station to the same one.

Possible solution:

The module should check if the new control voltage (coming either from the Station pot or Tune CV input) is going to actually change the currently played station to a different one and only then go ahead with the change.

TomWhitwell, 2015-02-15:
That is exactly the way that the system is coded to work, and I think it does work when you have a 3-4 files in the sound bank - you may get odd behaviour with only one file. 

    if (channel == PLAY_CHANNEL) CHAN_CHANGED = false; // Do not retrigger if channel is unchanged 

Could you test it to ensure that every time the pot is changed (i.e. 1024 times in a rotation) the bank is retriggering stuttering? 

That said, the checkInterface() function could do with a clean up / refactor / simplify - I noticed while checking this that there are 3 different variables called CHAN_CHANGED, channelChange and chanChanged, which is asking for trouble

PawelGrudzien, 2015-02-15:
Just had a quick look. It seems that I can replicate the issue regardless whether there is 1 or more files in the bank. When I turn the channel knob fast enough there seems to be a lot of stuttering going on. When I turn it from left to right (or right to left) very slowly (maybe over 15 sec.) there is no stuttering.

You can hear it very clearly especially with your test tone bank. If you wiggle the pot left and right quickly within the range of the 440Hz sine you will hear a lot of stutter, but if you turn it really slowly, there is no stutter at all (i.e. it works as you intended).

TomWhitwell, 2015-02-19:
Further example of this: 

I get a lot of crackle/jitter from the Station pot and CV. On the test folder, it sounds like this when wiggled by hand: 

[s]https://soundcloud.com/mqtthiqs/0002-4-audio-1/s-XBebi?in=mqtthiqs/ sets/radio-music-issue/s-dhaSm[/s] 

... and this is modulated by a LFO: 

[s]https://soundcloud.com/mqtthiqs/0003-4-audio/s-8f3qh?in=mqtthiqs/se ts/radio-music-issue/s-dhaSm[/s] 

When not modulated/wiggled, it's very stable and doesn't crackle.

TomWhitwell, 2015-02-19:
Closing this, but adding a link to issue #55

commit 1ad31ed0 (2015-03-03): Refactored Checkinterface()

Fixes #55
Fixes #49
Fixes #63
Fixes #51
closed 2015-02-19 by TomWhitwell