TomWhitwell/RadioMusic #52: Some audio files play in really low pitches.

repoman, 2015-02-13:
I used Max sbooth to batch convert a bunch of files. Some of them play in a really low pitch, not like the original sound at all.

TomWhitwell, 2015-02-13:
Sorry to hear that, how annoying. 

I'm 99% sure this is because you've converted files that were not recorded at 44.1khz. 
If you convert a 96k into a 44k .raw file without resampling, it will play at less than half speed - ie more than an octave lower. Similarly, a 22k file will sound like Mikey Mouse. 

I'll copy this answer to the FAQ and close this ticket - please reopen it if it doesn't solve the problem.

closed 2015-02-13 by TomWhitwell

repoman, 2015-02-13:
Turned out the reason was that they were stereo. Works just fine now. Such a fun module :)