TomWhitwell/RadioMusic #53: A different way to do bank selection

TomWhitwell, 2015-02-13:
Eclectics write: I find the press and hold reset to change bank a bit fiddly. Would be nice if you could press and hold reset, then use the pot to get the bank you want. 

Interesting idea, I'm not sure exactly how it would work - if you pressed and held, and the Pot was in a different position, you'd jump to that bank. Maybe one of those turn/catch modes, like on some analog synths with patch memory. 

Feedback?

jasondbaker, 2015-02-28:
The challenge with the existing bank change mode is that you need to time the hold just right otherwise you end up incrementing past your intended target bank.  Another option is to use a press-and-hold for 2 seconds to enable the bank change mode. The mode is signified by a steadily lit reset LED. Then you can press the reset button multiple times to increment through the banks. Press and hold the reset button again for 2 seconds to disable the bank change mode. You could still maintain the current behavior by allowing the user to press and hold the reset button without releasing the hold for a longer period of time.

mqtthiqs, 2015-02-28:
I like Jason's idea excellent! +1 on this.

clarke68, 2015-07-13:
The current way bank selection works is the only thing I don't like about the module. Another vote for changing to the way Jason suggests above...I'd love to see that on the next firmware update.

jamescarruthers, 2015-07-13:
I've been working on some pot code to manage this. One way is to make the pot position relative once you press the reset button. If you're on bank 2 but the pot is in position 14 then turning it CW would move it to position 4 at most. Turning it all the way CCW would get you to position 12.

jasondbaker, 2015-07-13:
Endless encoders would provide a better user experience than pots for this function. I appreciate the effort to resolve this issue. A solution based on relative positioning would create two challenges: a) moving the pot CW or CCW could produce similar results leading to user confusion, and b) the user would need to calculate which direction to turn a pot to achieve certain extreme values (i.e., 1 or 16). Another possible solution is to use a "snap" mode where turning the pot doesn't change the bank value until the pot value matches the current bank value. After the pot value snaps to the current bank value, future pot value changes will affect the value. This solution has user experience issues as well since users have to hunt for the current bank before changing it. I still think that modifying the existing button function offers the easiest solution.