TomWhitwell/RadioMusic #54: CV inputs noisy with module test sketch

mezzman, 2015-02-14:
Well I have ordered PCBs from Seeed and am having the EXACT same issue as the last build using PCBs from Itead.
Can anybody who has been building this module please run the test sketch and let me know if they are seeing any flickering of the CV input LEDs? I am going insane here trying to figure what is going on, I can't believe it is only me seeing this issue when I have the same problem on 10+ PCBs that I have put together already from different sources.
I need more help troubleshooting the possible cause as to why there is either noise or voltage leaking somewhere on the CV inputs.

TomWhitwell, 2015-02-15:
Are you having trouble with the actual module firmware, or just the test sketch? 

Many people have now successfully built the module with the Thonk kits.

Thonk-Steve, 2015-02-15:
A video would be helpful or pics of both PCBs

mezzman, 2015-02-15:
I have attached video of the test sketch and with the RM firmware running.
https://www.dropbox.com/sh/rjakcwng2j47rpl/AABLn1F8OFCWW7_TNH6-wnXia?dl=0
As you can see with a negative CV into the Start jack makes the noise stop. Otherwise it is as you see with no input. With the RM firmware running, the start flashes and output is jittery.
Also the trimmer pot on the back of the rear board is very sensitive, from zero to maybe a 1/10th of a turn turns up the volume then after that the audio cuts out.

mezzman, 2015-02-15:
There is also a 3rd video in that dropbox folder that illustrates another hum issue. When I touch the back solder pads of the TL074, the audio level jumps to proper level, than gradually fades in volume and gets distorted. One touch restores volume and then it fades back down again.

TomWhitwell, 2015-02-15:
Hi, 
What kind of power are you using? Is your output module in the same case as Radio Music? 

This looks like a grounding issue - I think that would account for the all the strange behaviour. 

Do you know anyone else with a eurorack case who could try the module for you? 

One thing to try - I'd be very surprised if it made a difference: tack a wire between the GND and AGND pins on the teensy (the pins indicated here: https://github.com/TomWhitwell/RadioMusic/wiki/Testing-and-calibrating-the-Radio-Music-module ) 

If your module is in a different case from the output module, try running a few patch cables between other modules in the two cases, see if that makes any difference (i.e. supplying more routes to ground)

mezzman, 2015-02-15:
The problem is the same with 2 different cases, one with a linear power supply and a Cell 90 with a switching PS. I have audio out of RM going straight to my external audio mixer, all equipment is on the same ground from the same power outlet. Trying different patch cables does not attenuate the hum.
I did try bridging GND and AGND and no difference.

TomWhitwell, 2015-02-15:
If you have audio going from the RM to another module (like a fully open filter or a mixer) in your rack, does it make a difference?

mezzman, 2015-02-26:
no difference. 
All I have to do is touch pin 12 of the TL074 and the hum goes away, all audio is fine. As soon As I take my finger away the hum fades back in over 3 seconds.