TomWhitwell/RadioMusic #55: Refactor checkInterface()

TomWhitwell, 2015-02-15:
The checkInterface() function could do with a clean up / refactor / simplify - I noticed while checking this that there are 3 different variables called CHAN_CHANGED, channelChange and chanChanged, which is asking for trouble

mxmxmx, 2015-02-16:
I thought this too, when looking at the code the other day. it also seems to read the ADC at fairly/unncessarily high rate (int checkFreq = 0)? that seems to be way faster than any change could take place (though i didn't read the code very thoroughly). i'd suggest using https://www.pjrc.com/teensy/td_timing_IntervalTimer.html

mxmxmx, 2015-02-19:
re #51 -- i vaguely remember the 'map' function can behave a bit funny sometimes. using the serial monitor, have you confirmed 'channel' is mapped correctly when FILE_COUNT[PLAY_BANK] = 1 (which i assume means there's one file, not two) ?

anyways, since the number is known, an alternative would be to pre-compute a/the mapping coefficient:

e.g.

float map[BANKS]; 

where map[num_Bank] = (float)FILE_COUNT[num_Bank]/1024.f;

then:

channel = (int)(channel*map[PLAY_BANK]);

if FILE_COUNT[PLAY_BANK] = 1, channel then should always map to zero if  0 < channel < 1024, so it's hard to see how turning the knob could make any difference.

PawelGrudzien, 2015-02-23:
re #51 -- I found the culprit!

In interface.ino, the line 
```
if (channel == PLAY_CHANNEL) CHAN_CHANGED = false; // Do not retrigger if channel is unchanged 
```
only gets executed if chanChanged > chanHystTime

But on every channel CV/pot change the variable CHAN_CHANGED gets reset to the default value from the settings file and if it is 1, then on a fast pot turn (chanChanged < chanHystTime) CHAN_CHANGED is set to 1 anyway - i.e. it will trigger the station change (and an audible click)

A simple fix is to change the existing code:
```
  if (channelChange && chanChanged > chanHystTime){
    // Change the channel global variable 
    channel = chanPot + chanCV; 
    channel = constrain(channel, 0, 1023);
    channel = map(channel,0,1024,0,FILE_COUNT[PLAY_BANK]); // Highest pot value = 1 above what's possible (ie 1023+1) and file count is one above the number of the last file (zero indexed)  
    if (channel == PLAY_CHANNEL) CHAN_CHANGED = false; // Do not retrigger if channel is unchanged 
    PLAY_CHANNEL = channel;
    channelChange = false; 
    chanChanged = 0;
  }
```

to

```
if (chanPotChange > chanHyst || chanCVChange > chanHyst) {
    // Change the channel global variable 
    channel = chanPot + chanCV; 
    channel = constrain(channel, 0, 1023);
    channel = map(channel,0,1024,0,FILE_COUNT[PLAY_BANK]); // Highest pot value = 1 above what's possible (ie 1023+1) and file count is one above the number of the last file (zero indexed)  
    if (channel == PLAY_CHANNEL) CHAN_CHANGED = false; // Do not retrigger if channel is unchanged 
    }

  if (channelChange && chanChanged > chanHystTime){
    PLAY_CHANNEL = channel;
    channelChange = false; 
    chanChanged = 0;
  }
```

The above change fixes the issue #51 and it doesn't seem to have any noticable impact on performace.

PawelGrudzien, 2015-02-23:
Now if only someone could explain to me the correct steps, I can upload the new code to github.

PawelGrudzien, 2015-02-23:
After further testing the above fix seems to work fine for the Channel pot, but Channel CV changes still occasionally trigger the channel change (even with only 1 file in the bank).

TomWhitwell, 2015-02-25:
Thanks @PawelGrudzien, I will look at this.

TomWhitwell, 2015-02-28:
Notes on checkinterface() redesign: Requirements work in progress 

Inputs and global variables: 

analogRead(CHAN_POT_PIN) = Channel Pot Position 0-1023
analogRead(CHAN_CV_PIN) = Channel CV Level 0-1023 
analogRead(TIME_POT_PIN) = Start Pot Position 0-1023
analogRead(TIME_CV_PIN) = Time CV Level 0-1023 

sampleAverage = How many times to read the pins and average the results 

PLAY_CHANNEL = currently playing channel 
FILE_COUNT[PLAY_BANK] = number of files in the current bank - ie potential channels 

Outputs + Global variables: 

NEXT_CHANNEL = Channel selected by interface; move to this channel the next time you do a restart

RESET_CHANGED =

TomWhitwell, 2015-02-28:
Thinking this through, I wonder the core of the problem is the PLAY_CHANNEL variable. 

I've been using this as both a way to discover what channel is currently playing AND as a carrier, a way to transmit channel changes from checkInterface() to the main loop.

TomWhitwell, 2015-03-02:
Re #63 - thinking the behaviour through 

Hitting Reset should always check the Start pot / Start CV and reset to the point selected.

TomWhitwell, 2015-03-03:
Testing for this refactor: 

- Can select all files in a directory 
- Being 'stuck between stations' doesn't sound horrific  
- Robust, no module crashes 
- Options in settings.txt all work 
- No glitches when moving station pot or CV 
- Reset goes to the same place, unless channel has changed 
- In non-looping mode, reset always goes to the same place regardless of channel changes 
- Looping & non looping works 
- Hot swapping works 
- No return of #3 or #37 or #28 
- Should solve #49 #63 #56 #51

commit 1ad31ed0 (2015-03-03): Refactored Checkinterface()

Fixes #55
Fixes #49
Fixes #63
Fixes #51
closed 2015-03-04 by TomWhitwell