TomWhitwell/RadioMusic #56: Strange behavior : flickering reset LED + stutter sound

Onosendaifr, 2015-02-16:
Hi,

Strange things begin to happend after 1 day of perfect working, and it seems to happend when the sound is stronger than a very little treshold : 

Reset LED begin too flicker and the sound stutter like a time streching effect (the sound continue to play but advance by grain as is the sound was, at high speed, stopped, then starting point moved a little bit further, then play resume).

I can't understand why, i've tried to reset the setup file without success. Any help ?

Thonk-Steve, 2015-02-16:
Do you mean this was a problem with a specific .raw file or part of a file?

Onosendaifr, 2015-02-16:
No, it happend with every files (i use you library).

It's very strange, because it seems to be trigerred by the volume of a sound, very quiet sound are played normaly, this behavior start when the level pass a certain amount.

The playback is ok when the module is "cold" and just booted the modular, but 3 or 4 minutes later, it begins.

Could be a bad soldering joint, maybe near the reset/start/station, i will double chack that.

Thonk-Steve, 2015-02-16:
If you power off your case then power on again once it happens.. does it still happen immediately? If so then if you power off, leave for 30 mins then power on again is it happening then? That might help rule out any temperature fluctuations that are expanding/contracting a cold solder somewhere.

The other thing to test is whether it happens when nothing else is plugged into the power supply, could you perhaps be near the limit of your power supply current? Radio Music does draw more power than your average 4hp module.

Onosendaifr, 2015-02-16:
> That might help rule out any temperature fluctuations that are expanding/contracting a cold solder somewhere.

It seems to be the case ! I will inspect than thanks.

TomWhitwell, 2015-02-19:
You could also check to see if this looks like some of the issues referenced in #55 re glitches while slowly moving the station knob - however, if you have sound stopping, that sounds very different

mezzman, 2015-02-19:
I have had the same issue and believe this is related to issue #54 
I believe there is a voltage offset present at the start CV which is causing the issue.
Onosendaifr, can you run the module test sketch and see if you have any LEDs lit without any CV inputs?

Onosendaifr, 2015-02-19:
What is the test sketch ?

I've inspected all the solder joint and correct the ugliest one, all my problems seem to be gone \o/ 
I will wait one or two days like this to see if it's ok, but as Steve find it, the reason was certainly a cold solder.

BUT

I also have the #55 issue :) Wich is a completly different problem i think

Do we need to wait for a firmware update about this one ?

TomWhitwell, 2015-02-19:
@Onosendaifr That's great news. I'll close this thread now, reopen if it comes back. 

#55 will be a firmware update, no ETA on that.

closed 2015-02-19 by TomWhitwell