TomWhitwell/RadioMusic #58: Resistor Fix on rev 1 boards

Thonk-Steve, 2015-02-20:
regarding this section of the build doc

* If you are using a Rev1 board, fix the grounding problem on R21 and R10 by soldering their legs to R13 at the 'super careful' end. This is already fixed on the red PCBs from Thonk.

Actually while this is indeed fixed on the red Thonk PCBs, I just realised they actually do say rev 1 still (the back PCB correctly says rev 2)

Other than that R21 and R10 ground trace fix on front pcb rev 1, are there other easy ways to identify which board you have if you don't know the source of them Tom?

TomWhitwell, 2015-02-20:
OH, how annoying. 

Yes: 

- New, updated boards have curved traces, old Rev1 boards have straight traces 
[image]

- New, updated boards have 1.5k LED resistors (like R11), old Rev1 boards have 1k LED resistors 
- You can see the fix in place on the new boards - R21, R13, R10 all connected to ground: 
[image]

I'll create an issue to fix the Rev. Number

Thonk-Steve, 2015-02-20:
Ah nice one cheers Tom

closed 2015-02-25 by TomWhitwell