TomWhitwell/RadioMusic #60: Reset doesn't reset the playhead to the same place

PawelGrudzien, 2015-02-24:
This one is easy to replicate.

1. Select a channel with a longer file (a long drum track is the best to demonstrate this issue)
2. Set the Start knob fully CCW and make sure there is nothing in the Start CV input.
3. Send a clock trigger to the Reset input (reasonably slow so it's easy to notice the issue - I set mine to 60BPM)

On each reset the playback restarts from a place slightly later in the file, whereas it should restart exactly from the beginning.

When you move the Start knob slightly and return to fully CCW position, the playback restarts to the very beginning and then it is affected by this issue again.

My settings.txt file contains MUTE=1 and DECLICK=3. I am yet to test whether this has any influence on this issue.

PawelGrudzien, 2015-02-24:
The files I can replicate it with are actually 4 seconds (and less) long.

I cannot replicate it with long files from the example sound pack - the playhead always restarts to the same place.

The MUTE setting doesn't seem to make any difference.

mxmxmx, 2015-02-24:
this might be due to lines 89-91 in Interface.ino ... ? 

looks as if file_offset is continuously added to "playhead", so "playhead" keeps advancing, whereas (I guess?) for one-shot purposes it should be reset to whatever would be returned by the Start CV+pot, ie when there's an interrupt. 

if i see it right (though i'm not sure what the purpose of playhead is), a quick hack might be to make "newTime" global (line 88) and reset the playhead to newTime when/if there's an interrupt.

TomWhitwell, 2015-02-25:
I just tested this, and unfortunately I couldn't reproduce it. 

I took 'BREAK1.RAW' and trimmed it to less than 4 seconds, resaved it and put it in a folder with the vinyl and birdsong. 

It behaved exactly as normal - every time I put a pulse into Reset, it started from the position selected by the knob. 

I tested it with Looping=0 and Looping=1

PawelGrudzien, 2015-02-25:
I just did some further testing and I can replicate the issue consistently.

It turns out to replicate the issue a variable CV needs to be send to the Channel input at the same time (to change between the files).

It is very easy to hear the issue using the following bank:

https://dl.dropboxusercontent.com/u/66094479/12.zip

This bank contains 48 files that are of identical structure: 1 second chord followed by approx. 3 sec silence.

Steps to replicate the issue:

2. Set the Start knob fully CCW and make sure there is nothing in the Start CV input.
3. Send a clock trigger to the Reset input (reasonably slow so it's easy to notice the issue - I set mine to 60BPM)
4. Send a slow variable CV to the Channel input. I send a CV from the turing machine output, clocked from the same clock as in step 2. (The idea is to have a random progression of chords, always played from the beginning of the file).

Over time you will notice that each file starts being played from slightly later in the file (on every Reset the playhead is slightly further, whereas it is supposed to be on 0). After about a minute you can't hear anything as the chords in my bank are followed by 3 sec. of silence.

If you wiggle the Start knob and set is back to fully CCW, the whole process starts over.

I can replicate the issue with both MUTE=0 and MUTE=1.

Hope the above is clear.

chapelierfou, 2015-03-01:
Hi, i am not sure if it's the same that happens to me : i have a SD loaded with drums one shots only. Looping=0. Simplest patch : a Pressure Points selecting various stations, the Touch output of brains going to the reset input. I assumed i could play drums with this. it looks like it's loosing the starting point as soon as i play a new station. I need to manually put it at zero to have the sample starting at the beginning.

mxmxmx, 2015-03-02:
as noted above, it looks as if file_offset (from play_sd_raw.h/cpp) is added to the "playhead" variable if/when the start position is changed. with every sample buffer, file_offset increases by 256 samples so "playhead" isn't in fact tracking the pot/CV position. as far as i can see, only file_offset is reset when the file is re-triggered but playhead is not.

i'd probably try adding (after line 238) something like:

else if (!Looping) playhead = newTime; 

you'd have to make newTime global first, of course (it's now local to the respective if-clause in Interface.ino). at any rate, it should be easy enough to ascertain things are ok (or not) by inserting Serial.println(playhead) after line 239 and watching the output on the serial monitor.

TomWhitwell, 2015-03-03:
Think I've resolved this in work on #55 

It definitely feels smoother and more logical. 

Essentially, 'looping=0' turns off the Radio metaphor completely. 

Still a bit of testing to do

Should have a beta out in the next few days, or you can browse my branch: 

https://github.com/TomWhitwell/RadioMusic/tree/refactorcheckInterface/RadioMusic

closed 2015-03-04 by TomWhitwell