TomWhitwell/RadioMusic #61: Refactor scanDirectory()

TomWhitwell, 2015-02-25:
At the moment, the module creates a directory of the entire SD card, copying filenames into an array.  

This, I think, is an unnecessary use of memory. 

Instead, the module could... 

1) Just load the current folder into memory 

or: 

2) Not load any filenames into memory, but use commands like openNextFile() and rewindDirectory()

TomWhitwell, 2015-03-09:
Making progress on this in [this fork](https://github.com/TomWhitwell/RadioMusic/tree/refactor-directory-scan), draft release notes: 

Fully reworked file system: 
- Can support unlimited files per SD card 
    - Each bank / folder can contain up to 500 files
    - Loading huge banks takes a long time, over a minute for 500 files 
    - The Reset LED flashes while loading a bank
- Can support up to 32 folders. When folders 16-31 are selected, the Reset LED is inverted (i.e. it stays lit and blinks off when reset) 
- Tested with files up to 2.5gb (8 hours) long. Scanning within the file takes about a second.

Thonk-Steve, 2017-03-15:
Does this one also affect the order in which files are structured on the Station knob?

TomWhitwell, 2017-03-15:
This fork has been completely superceded by Jouri's work, which re-wrote the file scanning & directory order system

closed 2017-03-15 by TomWhitwell