TomWhitwell/RadioMusic #66: RM stops working after a while

chapelierfou, 2015-03-19:
using the latest beta, non-looping mode.
i sequence rhythms from gates produced by a Trigger Riot, hitting the Reset input. After sometimes it stops working, and simply manually pressing Reset or replugging the jack (both ends work) makes it work again.
If i plug the TR's Gate out in a buff mult at first, it works consistently.

chapelierfou, 2015-03-30:
Any thoughts about this one ?

TomWhitwell, 2015-03-30:
Sorry, haven't had any other similar reports, haven't seen it myself. Not sure what to suggest. Did you ever get this behaviour with the old firmware? 

I think there may be an issue with changing banks - changing to a bank with different number of files gets confused - issue #37 fixed this with the old firmware - could that be it?

chapelierfou, 2015-03-30:
Well,  since when going through a buffered multiple, it works as it should, it sounds as a hardware problem. 
But in the other hand, i cannot remember of this in the previous firmware. Maybe it's worth a try. 
Regarding banks, i son't change baks, and it happened at a time when there was only one bank in my sd card.

TomWhitwell, 2015-05-13:
Hi, have you had any progress with this?

chapelierfou, 2015-05-13:
Nope. I didn't have the time to try with the previous firmware, though.

chapelierfou, 2015-06-27:
From Tiptop Audio : The RM probably needs to have a resistor to ground added.  Lots of modules do not have this even though it is critical.  THe 4ms stuff is all designed backwards with the resistor on the output.  It's a common design flaw in Euro.

Does it make sense ?