TomWhitwell/RadioMusic #69: RM doesn't work

zonkt, 2015-04-21:
Hello, after some e-mail exchange with Steve at Thonk. This one asked me to post my problem here. 
 When I connect it on the busboard nothing happens,
No red lights that comes on, only the orange led on the teensy on the back is flashing.
The teensy is well recognized when I connect it on the computer using the teensy software.
I start to mount the module with the teensy before install the firmeware.
I well check all my solders and everything seems to be clean.
I wonder if i get a right tennsy. Any ideas.?
If it's can help i have severals photos that steve asked me to send.
Thanks in advance.
all the best.
laurent

TomWhitwell, 2015-04-21:
"only the orange led on the teensy on the back is flashing" almost certainly means the Radio Music firmware is not installed. 
Check: 
https://github.com/TomWhitwell/RadioMusic/wiki/Preparing-the-Teensy-3.1
https://www.pjrc.com/teensy/troubleshoot.html

Closing this now - please reopen if you still have a problem

closed 2015-04-21 by TomWhitwell

zonkt, 2015-04-23:
Hi Tom,

 

Thanks for your reply.

I mount the module before installed firmware.

But didn’t soldered directly the teensy on the module

But on the female pin sockets.

So, after finished to mount the module, I installed firmware

In removing the teensy from the sockets and made it in connected

Directly the teensy on the usb computer.

So, I installed firmware, but should I do it in using my modular power

And cut up the USB power trace?

Thanks for your reply.

 

All the best.

 

laurent

 

https://github.com/TomWhitwell/RadioMusic/wiki/Preparing-the-Teensy-3.1
https://www.pjrc.com/teensy/troubleshoot.html

jamescarruthers, 2015-04-23:
If you cut the USB trace then you need to power the Teensy via the module — so connect the USB whilst it's in your rack and powered up.

zonkt, 2015-04-23:
Yes, it’s what i understood. But like I already installed firmware and made it again several times in seeing the module not working,

My question is do you think I should use this way to have my teensy and module working?

jamescarruthers, 2015-04-23:
I am not sure I totally understand.

Yes, you should cut the power trace as per the instructions.

Yes, you should load the firmware on to the Teensy whilst it's powered by the rack.

zonkt, 2015-04-24:
Oh sorry !

I just wonder that since I have not soldered directly the  teensy on the  module, and therefore I can remove it from its base, why should I do the firmware by connecting the module with the teensy on my system and not remove it from the module and connect it directly on the computer through the USB socket. I hope i'm a little clearer.

jamescarruthers, 2015-04-24:
Because it won't work unless it's in the module - because you cut the trace.

zonkt, 2015-04-24:
No, i didn’t cut the trace……Not yet!

It’s why I ask before doing it!

TomWhitwell, 2015-04-24:
OK, if you didn't cut the trace, just connect the teensy to your computer and upload the firmware. 

You only need to the cut the trace if you will ever want to have computer + modular power AT THE SAME TIME (i.e. because you are debugging the module or writing code).

zonkt, 2015-04-24:
Ok, it’s well what i understood !

I already upload the firmware….may be 10 times!

This one seems to be ok, it’s why I don’t understand it’s not working!

I can send you photos I made for steve if it’s can help!

TomWhitwell, 2015-04-24:
If the LED on the Teensy PCB is flashing, then you haven't installed the Radio Music firmware, you have installed the test 'blink' firmware.

zonkt, 2015-04-24:
Here is what i use to install.

zonkt, 2015-04-27:
Hi Tom,

 

Yesterday, I reinstalled for the thirtieth time at least teensy firmware, I installed it in my system and grrrrrr !!! I realize I have plugged it in backwards !! First time it happens to me. I turned, and taken off my hot radio music, I reconnected to the place! Ouff not grilled but still my two LEDs that flash back. I disconnected, I reload the firmware by pressing reboot, something I had also done many times! I reconnected in my system, and it works ...... red LEDs that light up front for the first time. When module’s world starts flirting with metaphysics, do not try to understand! J

Thanks for your help!

All the best.

 

laurent

Thonk-Steve, 2015-04-27:
Hi Laurent, 

Glad it's working! Do you mean you plugged in the power cable backwards or the teensy?

Regards

Steve

zonkt, 2015-04-27:
Hi Steve,

 

Thanks !

I plugged the power cable backwards !

It seems that the radio music loved that!

All the best.

 

laurent

Thonk-Steve, 2015-04-27:
Ah well that's a good test of the reverse power protection :)