TomWhitwell/RadioMusic #70: Distortion after module has been playing for a long period

TomWhitwell, 2015-05-13:
Celestial Tomes [writes](https://www.muffwiggler.com/forum/viewtopic.php?p=1895781#1895781): "I'm having an issue with the audio quality slowly becoming distorted over time. When I turn on my system the module sounds fine. Over the course of maybe 20-30 minutes the sound quality decreases."

I also had this issue - but over a much longer timeframe - leaving the modular on overnight. 

I think the problem is that there is no path to ground for the op amp input (IC4D) 
Last night I added adding a 330K resistor in parallel to C4. 

Based on one night's testing (!) this solved the problem completely. 

I'm going to continue experimenting with this to see if this can also cure #50 - my first test with 330k seemed to improve it slightly, will try with a lower value resistor. 

[image]

TomWhitwell, 2015-05-13:
Update: 
A 100k resistor works fine in that slot. 
It does not fix #50 but seems to improve it

celestialtomes, 2015-05-13:
tried a 200 K resistor. the volume level was cut down quite a bit. will try a 100k tomorrow

TomWhitwell, 2015-05-13:
Are you sure you have it in the right spot? I didn't notice any output level drop - if there was a drop, I'd expect 100k would be worse than 200k.

celestialtomes, 2015-05-13:
I did it late at night so maybe i didnt read the color stripes right. Im gonna grab some resistors from the shop and get back to you.

Michael Bailey

celestialtomes, 2015-05-16:
yah def wasn't 200K. I got a 100K on there now and it sounds great. Now onto the ground hum issue.

RichardDevine, 2015-06-10:
Hello Tom, I am having the same issue with my two units. Did you replace with a 100k-ohm 1/4 watt or 1/2 watt carbon film resistor?

TomWhitwell, 2015-06-10:
@RichardDevine 1/4watt will be fine.

RichardDevine, 2015-06-10:
@TomWhitwell  thanks going to get them today, :-)

TomWhitwell, 2015-06-10:
@RichardDevine Cool - mail me tom.whitwell@gmail.com if you need any help

oblivionratula, 2016-09-01:
Should this fix still be required on a kit fresh from Thonk? Guessing so since PCB says Rev 2. Just built mine tonight and getting the degradation.  If needed, I wish this was in the build docs so I could have taken care of it while the iron was hot (e.g. https://github.com/TomWhitwell/RadioMusic/wiki/Building-the-Radio-Music-module ).  Anyway, cool module, thanks.

equilet, 2016-10-18:
I'm having a (seemingly) related issue whereby the gain simply jumps up to around 3x the perceived loudness, but after around 1min.  There is no gradual increase in amplitude.  Does anyone else have this particular symptom?  Happy to file a new issue for this if that is more appropriate.

Thonk-Steve, 2016-10-18:
@equilet was that from a Thonk kit or a self build or..?

Does the volume trimmer on the back work smoothly and as expected in that first minute?

equilet, 2017-01-23:
Thonk kit that I built.
I'll check on the trimmer, thanks

dwhacks, 2017-08-30:
I just built mine as a Chord Organ and within less then a minute it would get all frizzly and the sound would die except for some static. I added a 100k resistor as documented here and it ran great for nearly 2 hours, and I bet it would last longer but its time for bed!

QbpNogCYUGmaGPzD, 2017-11-17:
I guess that the extra resistor is still something that needs to be added? It is not entirely clear from all the documentation.. The v2.1 board is unclear....

oblivionratula, 2017-11-18:
You'll know relatively quickly whether you need it or not.  I just built a Thonk-fresh unit a couple months ago (my second) and still needed it, so I'd guess probably.

QbpNogCYUGmaGPzD, 2017-11-18:
It is indeed. The distortion comes after 30 or more minutes here. I also noticed the rev3.0 label on this issue, so it is probably on the next iteration that it's solved. Let me check if I can make a pull request for the wiki, not sure, so that it's in the build manual.

QbpNogCYUGmaGPzD, 2017-11-18:
Added a comment on the bottom of the build guide to address the issue with a link to this issue.

QbpNogCYUGmaGPzD, 2017-11-18:
Added a comment on the bottom of the build guide to address the issue with a link to this issue.
@oblivionratula  @TomWhitwell

titotunes, 2017-11-25:
I bought a used pre-built Radio Music that didn't have this fix and I added it myself. However it seems it was too late and the opamp was fried. The symptoms I was seeing were that the station and time CV inputs were working intermittently and causing glitches, swapping the opamp fixed it.

Takennam, 2018-01-28:
Hello, i bought a pre built unit and i am also getting the distorsion. I realised that the distorsion happens with the Chord Organ firmware. I the 100k resistor the solution to this problem?

oliverlyu, 2018-07-06:
Hey there, I have just finished building my new radio music module, this is the first time I have done this.

I am able to play the module for around a minute until the audio stops/sometimes fades away and then the main audio drops, then there is ground noise. I have to turn the PSU off, unplug the module and wait for a bit and plug it all back up for it to work fine for that one minute again. I have added a 100k Resistor to the other-side of C4 but still no luck I have gone over my soldering 5 times and re soldered lose connections, I am at the point where I have checked every connection 3 times. can anyone help?

Many Thanks,

Oliver

QbpNogCYUGmaGPzD, 2018-07-08:
That sounds like something different, are you sure there are no shorts between the power rails? I think you better address this issue on Muff :)

TomWhitwell, 2018-07-08:
@oliverlyu That certainly sounds different from the 'slow distortion' issue. When the audio stops - are the yellow fuses hot (careful when you touch them)

oliverlyu, 2018-07-08:
@TomWhitwell as a matter of fact they do Get quite warm. Do you think this is an issue on my part?

TomWhitwell, 2018-07-08:
@oliverlyu The fuses will blow when the module is drawing too much current - and the way they blow is that they get hot - internally, two contacts separate. Unplug the power, they cool down and the two contacts come back together. 
Under normal use, that should never happen, so it sounds like you might have a short somewhere that means the module is drawing too much power, but not quite enough that it blows immediately. Start by checking orientation of diodes & capacitors. 
It may be that one of the fuses is hot, and not the other - that will tell you (by looking at the schematic) whether it's the +12v or -12v that is overloading, which might help

oliverlyu, 2018-07-09:
@TomWhitwell Hey Tom,

after further inspection it is not the yellow fuses that are getting hot, its the TL074CN chip below it. this gets incredibly hot when power is turned on. I have the notch lined up with the notch on the pcb outline too.

TomWhitwell, 2018-07-09:
@oliverlyu OK, that also suggests something very unwell - the TL074 might get very slightly warm, but certainly shouldn't be 'hot' - it sounds very like what happens when a chip is reversed, but you've already tested that, so is maybe a short somewhere else. You could try another TL074, but in my experience faulty chips are very rare - unless this chip has been 'fried' somehow by being reverse-powered in the past.

JonPigrem, 2018-08-20:
Hi, just checking - I have recently bought this kit from thonk (red boards marked rev 2.1). Do I need to do this fix or has it been addressed in the current version?

JonPigrem, 2018-08-20:
@TomWhitwell thanks dude, will do it to be sure.

Peace & Love

JonPigrem, 2018-08-22:
Just to feed back - I have the red Rev 2.1 boards from Thonk. I added a 100k resistor as directed above and the audio quality is super stable, even over long periods of time. I've had it running today for almost 5 hours without turning off and the audio is as it was when I started for the day. Thanks @TomWhitwell

duckythescientist, 2021-02-09:
I was browsing the schematics when I noticed the lack of a DC path for the op-amp and then went browsing the issues to see if it was a known issue. Here's an Analog Devices appnote on avoiding common op-amp problems. This is first on their list :smile: Hopefully it's useful reference for anyone else doing things like this. https://www.analog.com/media/en/technical-documentation/application-notes/AN-937.pdf

NeilMartin, 2022-04-27:
@TomWhitwell I'm currently building this module and now realize I have the Thonk rev2 board (2014). Thonk are out of kits so I sourced the components using the BOM, but the BOM doesn't include the extra 100k resistor for this patch. It would have been very useful for the BOM to include the resistor as I would have had all of the necessary components to complete the module. Can you add it to benefit future builders? Thanks for designing an awesome module!

TomWhitwell, 2022-04-27:
@NeilMartin Good shout, updated

pbr-street-gang, 2026-09-06:
I have a similar fault on my unit. Audio quality is fine when first placed in the case, then degrades progressively over a few minutes (like 5), sounds like the audio is being down-sampled, plus intermittent dropouts. Removing the module from the case (still connected via ribbon cable) and moving it around restores clean playback after several minutes. Put back into the case, it's fine for about 5 minutes, then degrades again.

Tried different rails/rows in the case, same result each time. Also ruled out the SD card and firmware, tested with multiple cards, clean reformats, and known-good files, no change.

I'm not confident doing the resistor/cap fix myself. Has anyone here done this repair for someone else, or know of a repair service (UK-based ideally) who's familiar with this specific fix?