TomWhitwell/RadioMusic #71: problen with sounds in folders

Tjongan, 2015-06-01:
Hi

Im having some trouble with the radio music module.When I put sounds in the other maps it crashes (it just makes a whizzing sound )
and I can't change the stations.. but when I have all the sounds in one folder it works..

Any tips on how to fix this ???

love the the module by the way

best

Jean-Louis

TomWhitwell, 2015-06-01:
Have you followed all the instructions here: https://github.com/TomWhitwell/RadioMusic/wiki/SD-Card%3A-Format-%26-File-Structure

Tjongan, 2015-06-01:
yes I have . It works fine if I just have sounds in the first folder , but
if I copy to the next folder it crashes  ..

j-l

   - Jean-Louis Huhta
   -
   - HUHTA HOME STUDIO
   - http://www.fitdetroit.com/
   -
   - DUNGEON ACID
   - https://www.facebook.com/dungeonacid
   -
   - Book
   - DUNGEON ACID
   -
   - jo@klsbookings.de  <jo@klsbookings.de>
   - http://www.klsbookings.de/artists/dungeonacid/

TomWhitwell, 2015-06-01:
How many files are you using - if you have >300 files on the card, you may have problems

Tjongan, 2015-06-02:
No its  not that many at all , and not that long either

j-l

   - Jean-Louis Huhta
   -
   - HUHTA HOME STUDIO
   - http://www.fitdetroit.com/
   -
   - DUNGEON ACID
   - https://www.facebook.com/dungeonacid
   -
   - Book
   - DUNGEON ACID
   -
   - jo@klsbookings.de  <jo@klsbookings.de>
   - http://www.klsbookings.de/artists/dungeonacid/

Thonk-Steve, 2015-06-02:
Are you going to provide some more information Tjongan? Screenshot of folder structures? File sizes? Details of how you're converting etc?

Tjongan, 2015-06-02:
Hi . Ive been testing a bit with different files in more folders and stuff
and it seems to be working fine.
I have a question about the best way to trigger the different samples in a
folder. Sometimes it feels like it's not changing the samples acording to
the triggers. But doing it from the rene seq works a bit better..

Do you have a tips on that?

best

j-l

   - Jean-Louis Huhta
   -
   - HUHTA HOME STUDIO
   - http://www.fitdetroit.com/
   -
   - DUNGEON ACID
   - https://www.facebook.com/dungeonacid
   -
   - Book
   - DUNGEON ACID
   -
   - jo@klsbookings.de  <jo@klsbookings.de>
   - http://www.klsbookings.de/artists/dungeonacid/

TomWhitwell, 2015-06-02:
Re: changing the samples 

- Take a look at the settings.txt options. 
https://github.com/TomWhitwell/RadioMusic/wiki/Customise-your-module%3A-Editing-settings.txt

- Also, checking the scale of the signals coming in - Rene might work well because the voltages are quantized? 

I'll close this issue now, if it's working OK for you.

closed 2015-06-02 by TomWhitwell

Tjongan, 2015-06-02:
Thanks

j-l

   - Jean-Louis Huhta
   -
   - HUHTA HOME STUDIO
   - http://www.fitdetroit.com/
   -
   - DUNGEON ACID
   - https://www.facebook.com/dungeonacid
   -
   - Book
   - DUNGEON ACID
   -
   - jo@klsbookings.de  <jo@klsbookings.de>
   - http://www.klsbookings.de/artists/dungeonacid/