TomWhitwell/RadioMusic #72: Loud hiss on the output comes and goes

gorasou, 2015-06-14:
Hi

I have a really weird issue with my Radio Music. There is a really loud noise on the output that's even louder than the files themself. thing is that it's gone sometimes, giving me a perfectly clean signal. I really don't know what the problem could be or what to do as I'm not a experienced diy-er. it worked fine for a couple days until there was hiss for the first time. and now sometimes it's there, sometimes not...

I hope someone can help out, thank you in advance.

Marc

TomWhitwell, 2015-06-14:
Hi! 

Please read #50 and see if it is same problem. 

If it is definitely hiss (ie high frequency white noise) and not hum (ie 50/60hz buzzing) then please record the effect and upload a link here.

gorasou, 2015-06-14:
It's not the same problem unfortunately, I already tried that :(

here's an example for the noise. it worked a couple hours ago but now it's back on...

https://www.dropbox.com/s/gqzuvhagbebstu0/Radio%20Music%20Noise%20Probelm.wav?dl=0

TomWhitwell, 2015-06-14:
Thanks, I haven't heard that sort of problem from anyone before, particularly if it is intermittent. 

It sounds like a high-pitched hum, different from the mains hum that some people experience when working across cases, and different from the noise that a few people have experienced when adding very quiet samples. 

And - it is inconsistent, so unlikely to be a simple hardware problem within the actual module. 

Not sure what to advise, you could try: 
- checking all of the connections on the board
- trying to understand what causes it to come and go 
    - Do you have other bits of equipment that are sometimes plugged in? 
    - Lighting, or cooling or other electrical noise-generating devices nearby?

michaelhaeusler, 2015-06-24:
Hi, 

I am having the same problem. It is the same hum and clicking sound but it is much louder compared to the sound from the SD-Card. The humming is always there . When the SD-Card is removed it changes it's frequencies (1 octave higher) und and is shortly interrupted periodically (every two seconds). 

I checked the module with the module hardware test and everything works fine and no noise is present but as soon as I install the RadioMusic Firmware (v1.1) the humming is there again.

@gorasou Did you find a solution for the problem?

Thonk-Steve, 2015-06-25:
michaelhaeusler is this definitely the same problem do you think if your noise is not intermittant as gorasou reported? Does it sound identical to the soundclip that gorasou supplied?

michaelhaeusler, 2015-06-25:
Yes, only that the noise is louder compared to the sound. Apart from that the hum sound is exactly like the soundclip from gorasou only that I have an additional periodical clicking sound and it is always there. I can provide a soundclip this evening.

Thonk-Steve, 2015-06-25:
I'll admit I'm baffled too as I've never come across one like this. gorasou and michaelhaeusler, it's probably worth posting photos of both sides of both PCBs.

michaelhaeusler, 2015-06-25:
I'll will provide some photos tomorrow. Here are the noises - the first with an SD card the second without:

https://dl.dropboxusercontent.com/u/3398946/radio_music_noise.mp3
https://dl.dropboxusercontent.com/u/3398946/radio_music_noise_without_card.mp3

michaelhaeusler, 2015-08-16:
So I finally made it to make some photos of the PCBs. Unfortunately the problem still exists. I double checked all soldering but the noise is still there. As I said there is no noise when doing the module hardware test so it might have something to do with the card reader? 

Here are the photos:
https://dl.dropboxusercontent.com/u/3398946/IMG_2371.JPG
https://dl.dropboxusercontent.com/u/3398946/IMG_2372.JPG
https://dl.dropboxusercontent.com/u/3398946/IMG_2373.JPG
https://dl.dropboxusercontent.com/u/3398946/IMG_2374.JPG

breedx2, 2015-08-16:
One of the yellow capacitors next to the power bus header looks like it's poorly soldered from the top of the board, and from the bottom I can't even tell what's going on there -- looks like a gashed open hole.  Fortunately, it looks like the traces are just on the top...

I would definitely touch that up with some solder from the top and then do a continuity check to make sure it's going where you think it should be going.

TomWhitwell, 2015-09-20:
Closing after inactivity

closed 2015-09-20 by TomWhitwell

michaelhaeusler, 2015-12-18:
Sorry for my very late reply. I had re-checked all solderings back then as breedx2 had suggested and made the continuity check but without success. The loud noise is still there which makes the module unusable. I had given up after that because I had no idea how to continue. Is there anything else I can do? Is it maybe possible that one component isn't working correctly? As I mentioned the noise is not there when doing the module hardware test. So maybe the SD-Card reader is somehow faulty? I am grateful for any kind of help!

djamsia, 2016-02-04:
I recently set up a Radio station.
there is no silence when no sounds, there is always a ground noise.
also get a loud hum from the reset input. when it used as an output.
I have the pcb rev2.1