TomWhitwell/RadioMusic #73: Power Supply/Clock Noise Leaking into Audio of Other Modules

Dubka, 2015-06-16:
I've noticed a strange, cyclical noise leaking into the audio of other modules in the same rack even when nothing is connected to the Radio Music.

Removing the SD card from the Teensy stops the noise.

Powering the Radio Music from a separate case to the other modules stops the noise. 

Any audio and/or CV connections made to and from the Radio Music make no difference.

Power supply is a Cincon and Tiptop Studio Bus.

jbook808, 2016-11-09:
I'm having this same issue with a RM kit I recently assembled.

Were you ever able to fix this issue? I would love to get this unwanted noise out of my system asap!

NickShaffner, 2017-06-16:
Same here, any luck?

jbook808, 2017-06-20:
Nope, haven’t been able to fix the issue.

rossgrady, 2019-06-04:
I'm also having this issue, but I only just noticed it when I switched to a different output mixer, possibly one that's more susceptible to this kind of leakage through the power supply. Very confusing, since I've had my RM for nearly a year & haven't noticed it before!

Now that I have noticed it, I did some testing & noticed that it's also present on the output. It still occurs when I pull the RM out of my rack and power it from my Mutable module tester. If I power up the RM without the Teensy installed, it's not present. If I power it up with the Teensy installed but without an SD card, it comes and goes as the RM boots up & starts looking for a card.

I may install some different alternate firmware just out of curiosity, but I doubt that would fix the problem while still leaving the RM usable as an RM. 

If I get ambitious enough maybe I'll try ordering another Teensy just to see if it's some manufacturing variance on their end or something.