TomWhitwell/RadioMusic #75: Background files restart from beginning each time despite looping=1

ckbrooks, 2015-07-28:
I'm experiencing a problem with the station knob and the background play feature.
Setting text file is using defaults, and looping=1.

However, when changing stations, the files restart from the beginning each time as opposed to the intended behaviour. Everything else within the module is operating perfectly.

Any help or rectification ideas would be very useful and welcome.

ckbrooks, 2015-07-28:
On 28 Jul 2015, at 5:38 pm, Tom Whitwell <tom.whitwell@gmail.com> wrote:

That's a mystery, not heard anything like that at all - please could you copy that mail into the Github issue log you started, see if anyone else has any ideas?

Are you using the 'suggested audio' sounds or your own files? 

Does the Start CV input work as expected? 

Tom 

On 28 July 2015 at 17:32, Dr. Clive Brooks......
Hi,

Thanks very much indeed for coming back to me so soon. 

No nothing going into the module at all to test it. I'm manipulating all the knobs manually.

The file loops round at the end perfectly fine, and indeed, it responds to different reset/start combinations fine too.

What is interesting is that it will occasionally continue in the background when the station knob is turned from one file to another - but only perhaps once in every three to five times at most. Quite curious. 

I hope we can solve it as it is quite a key concept in simulating the radio music behaviour. I, sure it's something daft I'm doing!

Cordially

Clive

ckbrooks, 2015-07-30:
Just an update. The module works fine with the Gamelan set, so it must be to do with the Audacity encoding of my RAW files. Has anyone got any ideas? I'm following the standard procedure I think.... Convert to mono, 16 bit, headerless RAW etc.

TomWhitwell, 2015-09-20:
Closing after inactivity

closed 2015-09-20 by TomWhitwell