TomWhitwell/RadioMusic #76: Fried my teensy

Lektroid, 2015-08-20:
I just managed to drop my radio module down the back of the modular while it was powered up, the two PCBs separated and it burned out the teensy. I've ordered another, but want to make sure the voltages are correct going to the teensy before frying a second one... With the common probe of the meter grounded (to one of the leads of the input diodes), I get these results on the 78L05:
Input: -12V
GND: 9V
Output: 4.75V

The output looks ok, but seems a bit strange that I'm getting 9V at the Ground point of the Vreg, I would have assumed it shared a common ground? Can someone let me know if these voltages are correct?

Lektroid, 2015-08-20:
tried a new Vreg, same (or very similar) results. I have no teensy board fitted at this point (just checking everything is normal before the risk of frying another).

I noticed a strange voltage reading to the teensy when I probe common to Teensy Ground point, I'm getting -5V where there should be +5V...? 

Going from the other end of the PCB, I've checked voltages from the power connector to the opamp inputs, all checks out as it should.

Lektroid, 2015-08-20:
*facepalm* I had the probes backwards in my meter

TomWhitwell, 2015-08-20:
Have a look here re: testing the power supply voltages: https://github.com/TomWhitwell/RadioMusic/wiki/Testing-and-calibrating-the-Radio-Music-module#test-the-power-supply

closed 2015-09-20 by TomWhitwell