TomWhitwell/RadioMusic #77: DC offset issue.

phidelity, 2015-09-01:
I am new to github and issue/bug reporting so if I'm doing this entirely wrong, please bear with me and point me in the right direction.

I've built a batch of Radio Musics and I'm pretty consistently getting the same problem with close to 10 of them, which means the error is possibly in my builds with a wrong component (looked and tested and everything seems to be in the right place, orientation & values) or perhaps a bad batch or fried 78L05 regulators or some other component. 

The issue is this: when looking at the output on a o'Jones scope and the amplifier trimpot all the way down - the volume has a DC offset of a volt or two, as I turn the amplifier trimpot up this increases the DC offset until its above what eurorack can handle ( ~3-4 volts above 0v ) and the audio clips until it disappears.

If I take this same signal into a 4ms SISM and apply a manual negative DC offset adjusted to 0v the amplifier works perfectly. Does this give you any clue where I should begin trouble shooting? Is there a specific circuit which handles this aspect of the amplification?

thanks!
kris

TomWhitwell, 2015-09-01:
The Teensy DAC has an offset of 2-3 volts, which is blocked by C2 and C3. If you have got substantially wrong values there (maybe 100p rather than 100n?) or they are shortcircuited somehow, that might explain it?

TomWhitwell, 2015-09-20:
Closing after no update

closed 2015-09-20 by TomWhitwell