TomWhitwell/RadioMusic #78: Error in CV input processing circuits

TomWhitwell, 2015-09-20:
From mbartkow
I'm sorry, if this has been asked before, but I couldn't find a discussion about this issue using the forum search function. 

Anyway, I am wondering, why the CV inputs are implemented the way they are, i.e. going through R10+R9 to a follower, and then force clipped to 0-3.3V range through R1 and D1+D2. This same applies to R20+R21, and then R2+D3+D4. 

First of all, I can't see any reasonable explanation of putting a 22k resistor in series with the input of an opamp whose impedance is many Mohms. Secondly, force limiting through a 47 Ohm resistor puts a significant stress on the opamp should the CV excess 3.3V, which is quite likely to happen in a modular environment, as the voltages are in -10..10V range. 

I am suspecting that perhaps there is a mistake in the schematic, and R9+R10 should rather be employed in a voltage divider, that would reduce the 10V range to 10V*22k/(47k+22k)=3.18V range.

mbartkow, 2015-09-20:
[image]

mezzman, 2015-09-30:
This may be the root of a lot of the CV issues I have been experiencing with my builds. If I mod the PCB and wire the resistors at the CV inputs as mbartkow illustrates, will this have any effect on how the module interprets incoming CV with the current firmware?

TomWhitwell, 2015-09-30:
@mezzman What issues are you having? 

I agree with @mbartkow analysis and it's probably worth changing in future, but think that the CV input is fairly accurately reflecting incoming voltages - that part has been tested quite extensively, and I've had no user issues reported. 
The firmware simply reads 0-3.3v voltages coming into the Teensy - so if you rewire the input, it should work in the same way without any firmware changes.

mezzman, 2015-09-30:
Issues I have encountered were Reset triggering while sending >5v into the Start input, and high pitch digital noise coming through other modules when CV is connected to the RM.

TomWhitwell, 2015-11-05:
Trying a PCB with this arrangement, but the resistances reversed (going through 22k, with 47k to ground) so that 5v is reduced to 3.4v - anything above that is clipped. I think the majority of use cases for this will be 0-5v - sequencer or LFO outputs. Easy enough to flip the values if you'd prefer 0-10v input.