TomWhitwell/RadioMusic #79: everything (might) work, except no audio out.

taylor12k, 2015-10-06:
i just built my 4th radio music. the previous 3 all went smoothly, it's not difficult. but on this one i'm not getting any audio out.

there appears to be normal activity on the LEDs and with CV input. i say "appears" because all i can do is go by visual activity... without being able to hear anything i can't tell for sure.

because i have 3 other working modules i've tried all of their SD cards in the new one, swapped cards around, etc... so i know it's not the card or formatting.

i've reflashed the firmware a couple of time and that seems to go fine.

when i first plugged the module into my case it was causing a weird power short ... so i took it out and realized i had a solder bridge.. once i cleared that up the power was fine.. but i'm hoping that didn't fry anything on the module. but, again, because LED activity seems to be "normal"... i don't think so.

any ideas?

TomWhitwell, 2015-10-06:
Things to try: 

Check the output pin from the Teensy - you need two rows of pins down each side, then another at the end - that's the DAC output, so if it's missing, it would behave as you describe, and it's the kind of mistake I'd make. 

https://github.com/TomWhitwell/RadioMusic/wiki/Building-the-Radio-Music-module#mounting-the-teensy

It's possible but unlikely that you could have fried the TL074 op amp - swap that with one another module and see if it makes a difference

taylor12k, 2015-10-06:
i swapped the op amp with a different radio music and it still doesn't work... but the other one i swapped it with does, so it's not the op amp.

the DAC pin is there...

TomWhitwell, 2015-10-06:
If you have other modules, you can swap the front and back boards with another - see if that isolates the problem - i.e. if the back board on your new module works on another, you know the problem is on the front board of the new module. (Hope that makes sense... )

Thonk-Steve, 2015-10-06:
was about to say exactly that :)

taylor12k, 2015-10-06:
good idea... will try that now

taylor12k, 2015-10-06:
yes, putting the back board from the un-working module onto a working one makes it not work.. so the problem lies in the back board.

Thonk-Steve, 2015-10-06:
Also swapping the teensy out? You need to test all three elements

taylor12k, 2015-10-06:
yeah, not looking forward to un-soldering two teensys..... nope!

Thonk-Steve, 2015-10-06:
ahhh... didn't realise you soldered them.

taylor12k, 2015-10-06:
hmmm.. time to throw in the towel?

TomWhitwell, 2015-10-06:
Assuming the Teensy is running code (as the LEDs suggest) the issue must be here: 
[image]
or here: 
[image]

Have you checked the trimmer is turned up? 
Checked the pins on the output socket? 
Have you remembered to solder R17 into place on the front board?

taylor12k, 2015-10-06:
hi tom,

yes, have moved the trimmer back and forth...

output pin seems AOK

don't see any missing resistor spots on the boards.. R17 looks good

Thonk-Steve, 2015-10-06:
Probably time to throw some photos up for your own sanity :)

taylor12k, 2015-10-06:
but then you'll see how atrocious my soldering skills are :)

taylor12k, 2015-10-06:
[image]
[image]
[image]
[image]
[image]

taylor12k, 2015-10-07:
did i scare you off with the photos? :)

TomWhitwell, 2015-10-07:
Gosh, when I look at photos like this I just think 'ouch, I could have routed that better' 

It all looks neat and correct to me - I think @Thonk-Steve has slightly more experience of photo debugging than me. 

We can't see the back of the TL074 - but unless there's something really obvious (unsoldered pins) I don't know what that would be. 
Unfortunately, the only RM I have at the moment is an earlier prototype - Russell Haswell stole my other working one.

jamescarruthers, 2015-10-07:
It's like cobblers children

taylor12k, 2015-10-07:
the back of the TL074 looks ok to me, no unsoldered pins or anything.... 
hmmm...

unless steve sees anything... might be time to find another one.

closed 2015-10-29 by TomWhitwell