TomWhitwell/RadioMusic #80: Blown teensy, now 3.3v line? Please help

adh82, 2015-10-26:
Hi,
I built the radio kit and all looks fine.
Before I plugged it in I was getting correct readings from the calibration instructions. I had it up and running then it just stopped, teensy got real hot.
Module is dead and it fried my  SD. Now when I plug the teensy into the computer via USB it gets really hot quickly but doesn't register with the app.
But after I spoke to STEVE from thonk about it he asked to check the 3.3v line and I'm not getting it.
Maybe around 0.12v?
Would anyone know what the problem is? Where to track the 3.3v source?
I want to get to the bottom of it before I install a new teensy.
Don't want another to die on me.

Cheers,

Adrien.

TomWhitwell, 2015-10-29:
Hi, sorry to hear that, 
Here are the instructions about testing the power supply circuit before you install the Teensy. 
https://github.com/TomWhitwell/RadioMusic/wiki/Testing-and-calibrating-the-Radio-Music-module

adh82, 2015-10-31:
Hi Tom,
So I've checked the voltages with your instructions and all is fine.
I do get about + and - 12.10v
And 5.10v
I guess this is due to my power supply. Is that ok?
Glad to hear the teensy outputs the 3.3v.
So in theory if I install a new teensy all should be ok regarding the voltages in getting?

Regards,
Adrien Harris

TomWhitwell, 2015-10-31:
Hi Adrien, yes - those voltages sound correct - don't worry about the 0.10v! Should be OK. 
Check your soldering when you added the pins to the Teensy - any shorts there could cause problems, or any nudging of the tiny components on the teensy itself

closed 2015-10-31 by TomWhitwell