TomWhitwell/RadioMusic #83: Text file with file order

poesboes, 2015-12-21:
Would it be possible to add a possibility to specify in which order files in a directory are to be played? I found that now files are ordered in file modification date/time order. If you now edit/replace a single file, it changes playback order and the last edited ends up last.
Thanks,
poesboes

jstenroos, 2016-03-22:
Sorting the files alphabetically after reading them in is one solution. You could then name the files using numerical prefixes "01xxxxxx.raw , 02xxxxxx.raw, ..." or similar. This way the order you see the files in explorer/finder would be the same as the play in the module.

poesboes, 2016-03-25:
Thanks for your comment. I'm afraid this is not what is happening. I've made a bunch of files representing MIDI note frequencies, named 000_C-1.RAW, 001_C#-1.RAW etc. I started off with a small subset of the middle range first for a trial and then extended to the full 127 note range. If i run across the set now, i first get the subset i created first, followed by the notes i created later. Files don't seem to get sorted by filename. Maybe it is sorted by way the entries are in the SD FAT? Then it would be a chance thing on the order of how your computer OS transfers files to the SD.

jstenroos, 2016-03-25:
You are right, it does not work like that now, I was just suggesting it as an alternate way to solve the problem. Currently it's the SD FAT order like you said.

I've done some changes to the firmware lately, and sorting is one of them. Maybe the changes will be incorporated to the official version at some point, meanwhile my version can be found in my forked copy  of the project here.

poesboes, 2016-03-25:
Thanks Jouni. I'll make a program for doing file sorting and forcing creation date setting before transfer to SD. I think it would save a lot of firmware space not having to implement file sorting?

jstenroos, 2016-03-25:
No, actually it's a memory friendly operation. I've run it with full 1200 files on the SD card with no problems.

poesboes, 2016-03-25:
Excellent! The more reason i'll have to dive into setting up the firmware programming of the system. Thanks.

closed 2016-03-25 by poesboes