TomWhitwell/RadioMusic #84: Power up problem

LeandroBarros, 2015-12-28:
I'm having quite a difficult time trying to find whats causing my Radio Music module to not power up at first. Sometimes i need to power cycle my case 2/3 times or maybe it's just totally random at this point, not sure anymore.

I've built the module with some Kingbright LED's, they're not super bright but they consume around 20ma.
I know the 78L05 can only provide 100ma for powering the Teensy, led's and uSD card reader and so i thought that was causing the problem... just soldered the leds that came with the Thonk kit but the problem still occurs randomly.

Things i've already tried and checked:
- Reinstalled RM firmware
- Install an alternate firmware that don't use the uSD card.
- Case psu's have enough mA available
- uSD card - SanDisk Ultra 16gb card class10 and it worked fine on other RM module i've built
- Put bypass cap across 3v3 and gnd (mxmxmx suggestion on Muff's) 

Mxmxmx also suggested  "give things a little more time: in the code, in the very first line in "void setup();" add, say, "delay(100);", recompile etc"

Already downloaded the files to build the modified FW but no clue on how to do this myself.

The module it's nicely soldered and the pcb it's also clean. 

Any idea where i should look? Maybe a bad Teensy? Thanks.

TomWhitwell, 2015-12-28:
If you have a good RM module, try swapping the front and back boards to find which one is faulty.

Thonk-Steve, 2015-12-28:
Does it help if you power on without the card in? Toms suggestion is definitely the way forward though.

LeandroBarros, 2015-12-29:
Not at all, same thing. 

I've built 2 RM modules, one always worked fine but this one always had some quirks.
After the first power up because the module had all the leds unlit but after a few power cycles led's were lit but had nothing coming from the output. Then somehow all seemed to be working fine but audio was very distorted even with the trimmer on the back set to full ccw. 
Checked and triple-checked every component to make sure everything was in the right spot, all good. Like always, also checked voltages on ic's sockets and teensy with my multimeter and all expected voltages were there, no shorts. 
It worked ok with my other main board so i reduced my troubleshooting to this board.  

Distorted outuput problem fixed by replacing the 074.. don't know how it got defective but that already happened to me quite a few times before with other projects. 

If i plug the teensy to computer, re-upload the firmware and then power the module in my case it always works at first, no random "power cycle" modulations :) 

If edit the settings on the card the module won't turn on at all. Deleting the settings.txt file only doesn't help, need to re-upload RM fw too.

Thanks Tom and Steve 

Cheers.

bassinfected, 2017-06-17:
Hi,
i have the same problem here with my Chord Organ.
Sometimes it powers up without problems, sometimes it needs a few trys but when it starts it works fine.
Anyone solved the problem already?

Best Marius

eeweegh, 2017-06-17:
Both my Chord Organ and Radio Music seem to boot up fine without the SD card inserted, so I use that workaround. Radio Music will reboot (without a problem) when it sees the SD card, Chord Organ will not react to the card insert (and work fine with its default settings), but the latest firmware allows you to reset it from the front panel (holding the button for >5s). So just hold that button after inserting your SD card.

Most threads where this problem is discussed, seem to converge on too much power being drawn at startup, when at the same time reading the SD card.