TomWhitwell/RadioMusic #85: Start CV input acting more like Tuner

rehills, 2015-12-29:
I am sending 0-5+ volts into the Start CV.  When I do, as I increase the voltage, RadioMusic switches files instead of moving the start position in the current file.  The Start knob works as it should.

Here are my settings:

MUTE=0
DECLICK=15
ShowMeter=0
meterHIDE=2000
ChanPotImmediate=0
ChanCVImmediate=0
StartPotImmediate=0
StartCVImmediate=0
StartCVDivider=64
Looping=0

rehills, 2015-12-29:
Results of further testing:  I removed my custom settings.txt file and the problem still exists.  Both CV inputs - Start and Tuner - act exactly the same.

TomWhitwell, 2015-12-29:
Do the knobs work correctly, but the CV works strangely?

rehills, 2015-12-29:
Exactly.  The knobs work as expected.  Both of the CV inputs act like the Tuner CV.

I got brave and pulled the RadioMusic apart.  I could see solder bridges on both ends of the BAT81 diodes below the reset button.  Looking at the traces, it looks like the diodes should be connected on one end but not the other.  I cleaned up the bridge but the problem still exists.  Probably a wild goose chase.

rehills, 2015-12-29:
I went over both boards with a fine tooth comb and I can not see any solder bridges.  The only two questionable connections were the diodes on the front board.  I cleaned them up so I know there are no issues there.  Any other ideas?

francescoco, 2016-06-21:
Refills, did you ever resolve your issue?