TomWhitwell/RadioMusic #90: High pitched whine

rjg700, 2016-02-28:
Having trouble with my newly built Radio Music, it's outputting nothing but a high pitched whine. Can't isolate an obvious fault in the build so would appreciate guidance as to what components would be likely to cause this. Already replaced the IC and electrolytic caps with no success. Have run Teensy tests fine and tried multiple SDs. C1 and R5 not touching. Cheers..

rjg700, 2016-02-28:
Also, I am almost certain this is not a grounding issue.

Thonk-Steve, 2016-02-28:
Have you tried it in a different case or on a different power supply?

rjg700, 2016-02-28:
I have tried in two separate Z-Rail cases with uZeus supply. Dont have a
second to try but it's definitely not overloading.

Thonk-Steve, 2016-02-28:
Have you tried with no other modules plugged in other than the Radio Music?

rjg700, 2016-02-28:
Hi, just tried this to no avail.

Thonk-Steve, 2016-03-01:
Do you have access to a power supply that isn't a uZeus just to check? They
are a little notorious for adding noise, although some people are running
them fine.

Where did you get your PCBs from? Was it a full kit?

rjg700, 2016-03-01:
Hi Steve, it's a full kit from you at Thonk. No access to alternate power
I'm afraid but I've never had any similar issues with the uZeus. Just to
clarify, I am not getting any sound at all from the unit apart from the
squeeling noise.

Thonk-Steve, 2016-03-01:
Do you know anyone else with a working Radio Music module?

What do the LEDs do?

rjg700, 2016-03-01:
No, unfortunately I don't. The LEDs seem to be functioning and look to be
metering the audio on the SD.

Thonk-Steve, 2016-03-01:
Ok the problem could be relatively simple then.
And if you hold the bank button you can then keep clicking and you see the
LEDs showing different banks? (they show it in a binary pattern - 0001=1,
0010=2, 0011=3 etc)

 I think the best step now then is to photograph both sides of both PCBs
and we can go from there. They need to be well-lit, in focus, square on so
I can both the components clearly and the soldering.

rjg700, 2016-03-01:
Yes, the bank selection is working correctly. Hopefully it is something
fairly simple! I have sent the PCB photos to your email, hope the lighting
is okay.

SkullandHeadphones, 2016-03-01:
Hi,

I'm not sure why I'm being copied on this email, so please kindly remove me
as I'm not sure what any of this is.

I thank you in advance.

Best,
T

Thonk-Steve, 2016-03-01:
I should be able to take a proper look at the pics in the morning thanks.

One thing I should have asked though, do you have a multimeter? have you
double checked on the power header that you don't have continuity between
+12v and GND... -12v and GND... or +12v and -12v ?

rjg700, 2016-03-01:
Cheers Steve. I don't currently have a multimeter to try but would it imply
that it might be worth trying a new power header?

Thonk-Steve, 2016-03-01:
Nope, it could be anywhere on any of the boards.
It's basically just one of the easiest and most common short circuit faults
to identify, worth borrowing a meter from someone.

protrolium, 2016-09-28:
I also get a high-pitch glitchy whirl after a couple minutes of use … oddly enough, briefly touching/tapping on the back of the module snaps it out of it.