TomWhitwell/RadioMusic #91: Issue with Start

Uebelwind, 2016-03-17:
Hi,

I just finished my first solder project and it surprisingly works almost perfect, but for one issue:

The Start knob doesn't work at all, neither does the Start CV input. So I assume there is an issue with either the connection between the two PBCs or between PBC 2 and the Teensy. 

I'd be grateful for an advise which pins are related to the Start parameter, so that locating the defect can be narrowed down.

Alex

TomWhitwell, 2016-03-17:
Have you read this page on the wiki: 
https://github.com/TomWhitwell/RadioMusic/wiki/How-to-use-the-Radio-Music-module#start-knob

Uebelwind, 2016-03-17:
Hi Tom,

thank you for your quick response. I will check this as soon as possible. Now that I've read it thoroughly, it make sense.

closed 2016-03-17 by TomWhitwell