TomWhitwell/RadioMusic #92: Reset CV input not working?

jamesbernard, 2016-03-24:
The module is working perfectly otherwise, Reset button works. But no matter what signal I send to the Reset CV input.. it never resets. I tried a number of different sources (MN Function, Doepfer A-160/161, Pulse clock).. no luck.

Any clues?

TomWhitwell, 2016-03-24:
Reset input is normally very reliable - it's a passive connection to the board via a couple of resistors. 
So - it can only be a soldering / component issue. 
On Front board (with pots @ sockets) 
Check r19 and r18 are what they should be 
Check d7 and d8 are the right way around 
Check inter-board connecting pins are all soldered 
Check all teensy pins are soldered correctly 
Check the reset jack is soldered correctly 

That's all it can be, really - probably either a missing component (r18 absent or swapped with a higher value would do it) or a short to ground somewhere

jamesbernard, 2016-03-25:
Checked all solder points and did not find any issues or ground loops. What finally worked was removing the 220R (R18) and placing a jumper in it's place.

closed 2016-03-25 by jamesbernard