TomWhitwell/RadioMusic #93: Resetting continuously by itself

Setherian, 2016-05-13:
I just finished my module, started testing and the reset function is being activated by itself continuously in a unrecognizable pattern (random?) (led blinks accordingly), its a very fast process and I dont know how fast it should be in terms of actual sound response but in the end I either get a recognizable reset or "clicks". 
What to do?

TomWhitwell, 2016-05-13:
If it resets regularly every 400ms, then you have probably soldered the button 90 degrees off where it should be - so it looks electrically like it's constantly pushed down. 

If the resetting is much more random/faster than that, it might possibly be a grounding issue - try the module in a different case, patch a few other cables into the module. 

Reopen this case if neither of these sounds useful.

closed 2016-05-13 by TomWhitwell

Setherian, 2016-05-13:
Hi Tom, after extended testing, the self reset frequency varies, but usually it takes a few seconds to appear and sometimes it can not happen at all for a long time, but the random burst of clicks continue and very frequently so, they seem to clearly relate to the "above reset LED" blinking, I think you are right about ground , while the behaviour didn't changed when I switched cases, it changed a bit, only for a while though when I sent the output straight to my audio interface, instead of going to other modules. SO yeah for a few seconds it seemed like the blinking n clicking was way less regular but still happening sometimes but now even in that config, straight to computer I get frequent blinks n clicks.
But I have no idea how to deal with this grounding if that is indeed the problem, what do you suggest?