TomWhitwell/RadioMusic #97: Radio Music won't start looping at power up unless I unpatch a few things

antikreon, 2016-05-27:
I have a Synthrotek Deluxe power module powering a MTM Radio Music module in an smallish 84hp case. My Radio Music won't start looping unless I unpatch my Bubble uLFO from my Dixie II Plus before powering everything on. The Radio Music module is getting power, regardless of how everything is patched, but I suspect a slight voltage drop caused by the patched LFO and VCO is causing the Teensy module on the Radio Music not to boot up. Any one seen this or have a solution? I would try bypassing the bus board the radio Music putting power cable directly into one of the power module's two available 12 pin headers, but my guess is that the electrons won't be fooled.

TomWhitwell, 2016-05-27:
That's very strange. The module has a voltage regulator on board, so small changes in the +12v line shouldn't make any difference. Have you been able to test the module in another case?

antikreon, 2016-05-27:
I haven't. It works great after unpatching. if it doesn't kick on I can hear the slight click when I remove the SD card and push it back in, so I know it is getting power. I had another similar issue I resolved: I connected two DC barrel connectors to my bus board to power my external Moog Werkstatt o1 and Mother 32. The RM wouldn't start if those externals were powered at the same time. I resolved that by adding a toggle switch to those connectors and switching them on after the main power is up. I'll try bypassing the busboard (it's loaded with capacitors to eliminate noise) and see if that works any unexpected magic. I'm on a island where I live an far as modular synth users go. I don't have access to another case (yet). I see the voltage regulator is accessible from the back board. Does it output 5 volts? Might checking the output voltage and possibly replacing that help?

antikreon, 2016-05-30:
Direct into the PSU didn't make a difference. I have found that powering up the RM with the SD card out helps. Using a 1 GB Sandisk card.

JonathanButler, 2016-07-31:
I just built my module yesterday and I have the same issue.  The unit will intermittently boot up in my Synthrotek Skiff.  It seems fine if I take the SD cart out, reboot, and put the card back in.