TomWhitwell/RadioMusic #98: station knob and tune cv not working

francescoco, 2016-06-19:
Hello,
Station knob and tune cv input do nothing. Have checked the pcbs several times for lifted traces or solder bridges on those paths and all seems ok. Resistor values are correct.
I did have a lifted trace between the anode of led 1 to pin 2 of JP2 which I fixed.
I'm wondering if there isn't something wrong with the card structure because I can only play one file from each bank (at best I can change files once if I turn the knob counterclockwise). Card formatted FAT32. Have tried several cards but problem persists.
Any idea where I should check next?

TomWhitwell, 2016-06-19:
Have you tried the Empty formatted card with scratches and birdsong? 
https://github.com/TomWhitwell/RadioMusic/wiki/SD-Card%3A-Format-%26-File-Structure

francescoco, 2016-06-19:
yes I did, in each bank I only hear the vinyl file.

----- Mail original -----

https://github.com/TomWhitwell/RadioMusic/wiki/SD-Card%3A-Format-%26-File-Structure

francescoco, 2016-06-21:
I tested voltages on the channel and start pots as well as on the teensy pins for channel and start cv while turning the pots. Voltages change, nothing shorting to ground.  Any idea where I might look next or where the problem might be coming from?

francescoco, 2016-06-23:
Went over the whole circuit checking for lifted traces, cold solder joints, solder bridges, component values. Everything appears ok with the hardware. Has anyone experienced problems when formatting sd cards on mac with El Capitan? For some reason I can only access one file per bank.

francescoco, 2016-06-26:
Final question before I order a new kit :  the leds I used are of the following specs 
voltage :  2 V
power consumption : 20mA
 Luminosity : 1600 mcd

I had to replace the kit leds because I did not follow the instructions and cut them too short :(

could they be the responsible for the channel pot/cv not responding?

francescoco, 2016-06-26:
Yes I have but the problem persists : only one file from each bank plays - in fact i can only hear the vinyl files if I leave both files in each bank; In bank 15 likewise, i can only hear the vinyl sounds;
Is there a pin on the sd cardholder that could be related to the problem? Or could it be the card structure?
I've checked over for lifted traces, solder bridges, incorrect component values and/or components incorrectly placed and am stumped:
I've formatted and reformatted one of two cards several times as fAT32 on mac OS.
Did the module hardware test; all seems normal.

----- Mail original -----

https://github.com/TomWhitwell/RadioMusic/wiki/SD-Card%3A-Format-%26-File-Structure

francescoco, 2016-07-11:
double checked connections between teensy and micro sd cardholder. card is a sands ultra 16GB. I'm at wit's end.  Anyone?

Thonk-Steve, 2016-07-11:
So is this perhaps an issue with the station pot or CV input rather than anything to do with the SD card? i.e. you're unable to tell it to move to the next file, rather than the card of file itself being an issue?

Is the pot showing a change in voltage across it's terminals when it's turned?

francescoco, 2016-07-11:
Yes pot shows voltage change across terminals and the corresponding pin on the teensy (pin 23) shows voltage change as well.
The Start pot works fine

francescoco, 2016-07-11:
Thanks for your reply:
Yes pot shows voltage change across terminals and the corresponding pin on the teensy (pin 23) shows voltage change as well.
The Start pot works fine

----- Mail original -----

TomWhitwell, 2016-07-11:
Have you tried with CV inputs - whether you can change the sample from birdsong by adding CV to the 'Station' input?

francescoco, 2016-07-11:
Yes, nothing happens.

Thonk-Steve, 2016-07-11:
francescoco where did you purchase it? was it a kit or just PCB or?

francescoco, 2016-07-11:
It was a kit from Thonk.

francescoco, 2016-07-11:
I did replace the leds however as mentioned in a  previous post

francescoco, 2016-07-11:
When I try to change the channel, the leds do not respond. They should light shouldn't they?
They were doing so before but don't now - in any event there was no station change (albeit occasionally it will change from one station to another only once but with no possibility of going back to the initial station . I've "de-" and "re-" soldered over the board so many times and replaced the headers with hardwiring so it's all getting quite ugly.

Thonk-Steve, 2016-07-11:
I'm thinking one option maybe that maybe the Teensy got damaged when you originally soldered the headers on, have you been removing the Teensy when performing rework?

francescoco, 2016-07-11:
No I can't say I remove it when performing rework. Is there a way I can test the Teensy to see if it's damaged?
If not, maybe I should just take a chance and reorder a teensy. 

----- Mail original -----

Thonk-Steve, 2016-07-12:
if you're seeing the voltage from the station pot and jack get all the way
to the teensy pin then yeah I guess it could suggest the teensy has been
damaged. To be honest that's outside my expertise now, maybe there is
software you can load to help you monitor pins.

But if you are reworking you should definitely remove both ICs and
processors like the Teensy, the excessive heat periods of desoldering and
rework can damage them quite easily.

francescoco, 2016-07-12:
Thanks for your help. I'll try another teensy and confirm whether it is indeed the root of the problem. 

----- Mail original -----

francescoco, 2016-07-31:
Hi,

Re: [TomWhitwell/RadioMusic] station knob and tune cv not responding (#98)
Have just installed a brand new teensy 3.2 and the problem persists : impossible to change channels whether I use the cv input or the pot.
I guess I'll have to order another radio music, but is it possible to get the kit without the teensy board?

cheers

----- Mail original -----
De: finizio@free.fr
À: "TomWhitwell/RadioMusic" <reply@reply.github.com>
Envoyé: Mardi 12 Juillet 2016 15:24:13
Objet: Re: [TomWhitwell/RadioMusic] station knob and tune cv not responding (#98)

Thanks for your help. I'll try another teensy and confirm whether it is indeed the root of the problem. 

----- Mail original -----