TomWhitwell/RadioMusic #99: Radio Music not always powering on.

zanscath, 2016-06-23:
So, I built my module almost half a year ago. It worked from the very beginning and never had any problems. I first used it with uZeus Power, then with power from a Doepfer DIY kit, general power needs are way below maximum available. Since I moved the module to the Doepfer rack it sometimes doesn't start from the get go. If I reset the power it usually starts immediately and works perfectly until I power off the system. Sometimes it starts without this trick, other times it doesn't.

Any idea why this could be? It's the only module in my rack that behaves like this though it's not the only one based on a MCU (don't know if this is relevant).

Thonk-Steve, 2016-06-23:
Is it reliable if you don't put the SD card in until immediately after you
power on?

zanscath, 2016-06-23:
I shall try over the course of a few days and get back with an answer. I did try three times right now and it worked every time, but I wouldn't say I'm out of the woods yet because the system was just online. Will get back to you ASAP.

zanscath, 2016-07-05:
Yup, seems to be a pretty reliable solution, thonk you!

kallepau, 2017-01-02:
I have the same issue. The RM doesn't always start up, but generally does after a few on/off switches. ("Have you tried turning it off and on again" comes to mind). 

Will try the suggested solution of putting the SD card in right after power on.

Thonk-Steve, 2017-01-03:
let us know @kallepau

kallepau, 2017-01-04:
Unfortunately, that didn't work. For now resorting to on/off switching.

Any other pointers for troubleshooting?

eeweegh, 2017-04-14:
similar observation with a Chord Organ I just build: it started up every time (say 4-5 times) with my test power supply (a Doepfer beauty case). Then I put it next to the Radio Music in a Doepfer LC6 case (PSU3), and now *both* refused to power up, even after I removed cards. After two or three attempts, it worked.
Radio Music resets when inserting an SD card, but Chord Organ (beta 0.5) does not, meaning I lose the  settings that I prepared on the SD card (like extra wave banks).

Thonk-Steve, 2017-04-14:
Anyone else using PSU3? I know there is a growing list of digital modules now which the PSU3 doesn't play nicely with, unfortunately it's getting a bit notorious.

eeweegh, 2017-04-14:
Well, I've had problems running a 4ms SMR on the PSU3 5v supply (12v was ok, so not really a problem), but otherwise the power seems fine for the rest of the modules in that case. Note that I had the same startup problem when running the Radio Music on an external Doepfer PSU (the ones that come with the DIY kit). Likely linked to what other modules draw at power up.

jamescarruthers, 2017-04-14:
Guessing you could test solve this by removing the power filtering e-caps

eeweegh, 2017-04-14:
> Guessing you could test solve this by removing the power filtering e-caps

Not at home in electronics, but they must be there for a reason, no?

jamescarruthers, 2017-04-14:
Power filtering - but it would work without them. The issue (I think) is
that every module has power filtering caps - so when you switch the power
supply on there's a huge strain whilst they all charge up. Pretty sure the
digital modules that aren't working are caused by the slow rising voltage.

eeweegh, 2017-04-14:
Thank you, I understand. Maybe that's the reason why it works after a few attempts, when the caps are still partially charged? I'll see if I can reasonably work around the problem (move some modules around, bring up the 6U case in two phases), but only attempt to remove the caps when I get really desperate (don't really want to run into other stability problems).

eeweegh, 2017-04-17:
while reading up on Teensy, I found out there's a reset pin (hidden as a pad on the back of the board). Does anybody have some suggestions about adding a momentary switch (to ground, I presume)? It sounds to me that it would allow to reset the Teensy (RM/CO) module individually after power has stabilised (which sounds more civilised than yanking a whole bus board of modules).

eeweegh, 2017-04-17:
After a few hours of testing, I found out that:
- emptying the rack with the PSU3 *completely*, except for the RM and the CO: problem is still there (cannot reliably start up the Teensy in either of them, regardless of card inserted or not). Maybe the theory of the power sucking caps still holds, but then these two modules are the only ones to blame (when filled, the case draws around 1A on 12v, .26A on -12v);
- removing either of them, makes the remaining one behave as mentioned above: boots up fine without SD card;
- so I ended up moving the CO to a different PSU, refilled the PSU3 rack, and everything now runs, assuming I take out the cards first (which is not really what I want on the OC, as mentioned).

Any suggestions to improve/fix this are still welcome. Adding the Teensy reset button I mentioned this morning is something I'll only venture when somebody pushes me :-)

oblivionratula, 2017-04-17:
On 04/17/2017 05:05 AM, Erik Wybouw wrote:
> while reading up on Teensy, I found out there's a reset pin (hidden as a
> pad on the back of the board). Does anybody have some suggestions about
> adding a momentary switch (to ground, I presume)? It [...]

This thread says pulse RST to ground to reset: 
https://forum.pjrc.com/threads/35269-Teensy-3-2-Reset-signal

This is a crack at the hardware issue of using that pad: 
http://rishifranklin.blogspot.com/2014/06/teensy-31-reset-pin.html

Not sure if that's usable on a Radio Music as I'm not near mine.

-- 
Mark Matula
http://Ratula.net

eeweegh, 2017-04-17:
Thank you! The first link cleanly confirms what I cobbled together from specs and other threads. The second one, I'd also come across, and I think it is possible (at least physically) to do on my Radio Music: I mounted the teensy with the female headers, so there's plenty of space. 
Once I get a wire off that reset pin, it's a matter of testing, then finding a neat solution for a button (preferably in the front panel, otherwise in a blank next to it).
But still hoping that somebody tells me that it's just a bad soldering or component - I just saw Steve's case for Superbooth, and it has at least 4 of them next to each other :-)

Thonk-Steve, 2017-04-17:
yeah we have 4, that's the thing that is confusing, also we have sold a LOT and there are only a tiny number of people who mention this so it's very hard to track down.

Is everyone using decent SD cards? Kingston or Sandisk?

eeweegh, 2017-04-17:
yup - brand new 16G sandisk in both of them.

Don't lose sleep over it - my OCD inner self wants to get this solved, but I'll find it something else to fret over :-). Just drop me a line if you come across an idea.

paulfreeman, 2019-04-15:
I also have this problem, with the latest firmware and a fresh Sandisk Ultra card.  I have the RM mounted in an Arturia RackBrute 6U which in addition to the RM holds a Behringer Neutron. The power on problem was happening before I put the Neutron into the case. 

It does seem to just be the first time, once it 'lights' the first time, then power cycles seem to work. Makes me wonder if its the big caps as there seems to be a 'priming' effect going on.

Thonk-Steve, 2019-04-15:
Paul does it boot more reliably if you leave the card out until after the
case is powered on?

paulfreeman, 2019-04-15:
Card in or out doesn't make any difference.  Its not a big issue, but thought it worth reporting in case there's some way to fix it.

NicolasRuaud, 2019-04-16:
Hi ,I build a RM last week and have the same issue. RM never lights up at first boot. Card in or out (fresh sandisk extreme 32Go) doesn't make any difference too.
 Alim is a doepfer PSU3. I check all resistor's values, all diodes and soldered points, all seems to be OK.
Does anybody try adding a WMD Soft Start Power Supply Delay ?

Thonk-Steve, 2019-04-16:
Can you describe more what the LEDs do on power on?

NicolasRuaud, 2019-04-16:
At first powering on, LEDs don't lights, there is no sound, the  module seems offline.

Thonk-Steve, 2019-04-16:
That's not the same issue described in this thread, if you're getting no
LEDs at all then potentially the Teensy isn't programmed or you have an
issue with the construction, if you bought the parts or PCBs from Thonk
then please email photos of both sides of both PCBs and the Teensy to
support@thonk.co.uk

NicolasRuaud, 2019-04-16:
Yes my issue is the same that the one described in this thread. My RM works perfectly but it do not always powering on. Sorry if I was missanderstood.

I was just wondering if a wmd soft  start could be a workaround.

paulfreeman, 2019-04-17:
I have fixed this issue on my board by re-flowing the solder on one of the electrolytic capacitors on the power input,  one of them had a dry joint, after tidying that up it powers up correctly.  @NicolasRuaud  may be worth checking the electrolytic capacitors.

NicolasRuaud, 2019-04-17:
Hi Paul, thank you for your answer, I am interressed with your solution, please can you post us a picture of your modification where we can see the soldering bridge to add?

paulfreeman, 2019-04-17:
There was no bridge, I just had to correctly solder the part. 

Look on the left, there is a bad solder joint. 
 
[image]

eeweegh, 2019-04-17:
FWIW, I bought a WMD Soft Start to see if it would bring any improvement (I am personally convinced it has to do with inrush current), and so far (one weekend) the combination has worked fine. Will post back in a few weeks if the solution remains stable, sooner if not :-)

NicolasRuaud, 2019-04-17:
Nice, good to know Erik!
 I'll check my electrolytic capacitors solders, and will try with a WMD soft start too.

NicolasRuaud, 2019-05-05:
FYI,  I bought a WMD Soft Start and its has also solve the problem for me.

eeweegh, 2019-05-11:
Confirmed after a few weeks usage, the WMD Soft Start has reliably solved my RM startup problems.

eeweegh, 2020-02-23:
You might be interested in this thread: https://forum.pjrc.com/threads/56541-Teensy-3-2-not-booting. In there, Paul Stoffregen explains how a power supply issue at startup could cause a Teensy to hang, and what was done in recent Teensyduino versions to fix that.
Found it by discussing with Befaco why a VCMC would sometimes fail to start, except when behind a WMD Soft Start.

head-monkey, 2020-04-15:
Hi. I had the issue of Radio Music not powering up in my Doepfer LC6 case with a PSU 3. I found that if I quickly turned the  power on, off and on again it would then power up with no problems.
The same Radio Music is now in a rack powered by a Tiptop uZeus and powers up first time with no issues whatsoever.

jkolhi, 2021-05-15:
> I have fixed this issue on my board by re-flowing the solder on one of the electrolytic capacitors on the power input, one of them had a dry joint, after tidying that up it powers up correctly. @NicolasRuaud may be worth checking the electrolytic capacitors.

This solved my problem also! Thanks so much for posting this! 
I did get the WMD Slow start module before testing this, and it did not solve my problem. Re-flowing the soldering points to the capacitors did the trick. Without the WMD module!

AbsoluteMediaSolns, 2021-08-04:
Been reluctant to chip in because I'm new to DIY having only completed five modules and with limited electronics knowledge. However I have the same issue. Radio Music works fine on my test PSU but when I install in the main case, powered by an Elby ED705, it rarely fires-up. I've reflowed the power caps but no difference so am considering the WMD soft start unless there are cheaper alternatives to Australia.

head-monkey, 2021-08-04:
> Been reluctant to chip in because I'm new to DIY having only completed five modules and with limited electronics knowledge. However I have the same issue. Radio Music works fine on my test PSU but [...]

I can't say for anyone else but I found that the WMD Soft Start made no difference to the start up problem with my Doepfer PSU.

jkolhi, 2021-08-05:
> I can't say for anyone else but I found that the WMD Soft Start made no difference to the start up problem with my Doepfer PSU.

I have the same experience: I got the WMD Soft Start but I have the same problem still. I have a Meanwell PSUs.

TonTonD, 2021-09-10:
I have two Radio Music's and I have been having this exact problem, but only with one of them. 

I decided to swap my two SD cards over, and the problem moved over with the card. So – for me at least – the problem must be with the SD card.

Thonk-Steve, 2021-09-10:
Hey,

That's interesting... What are the brands and models of the cards?

Regards

Steve
Thonk

TonTonD, 2021-09-10:
> Hey, That's interesting... What are the brands and models of the cards? Regards Steve Thonk
> […](#)

The one that is working perfectly in both: Gigastone 16gb micro SD HC.
The one with the problem seems to be unbranded, just says "2gb micro sd-c026 taiwian" on it.

TonTonD, 2021-09-10:
I've ordered a few different SD cards, so I'll try them out and see which ones make a difference.

Thonk-Steve, 2021-09-10:
Ok got it thanks, worth noting though, as per the build instructions we
only recommend Kingston and SanDisk cards.

Steve

TonTonD, 2021-09-10:
Ok, thanks. Yeah, I bought my Radio Music's second hand from eBay and these were just the SD cards in them. 
I'll replace with SanDisk. Thanks

steffenlohaus, 2021-11-02:
Hi guys, thnx for the contribution so far! I am experiencing the same problem, but am actually happy having found this thread 😅

My setup:
- Chord Organ DIY Kit Rev2 (the red one)
- SD card: SanDisk Ultra 32 GB
- PSU: Doepfer PSU3
- Digital module neighborhood: Bastl Popcorn (Sequencer)

I tested the workaround of pulling the SD card out when starting up. After ~5 attempts with a (hopefully completely uncharged system) it seems to fix the bug.

I would love to prevent or automate the workaround in the future.

That's why (@Thonk-Steve ):

1. (How) can the software be adjusted to maybe read the card after a certain delay?
2. (How) can an added capacitor at the cardholder help?

I thank you in advance!
Steffen