TomWhitwell/RadioMusic #101: Hi Friends,

SynDrum, 2016-07-22:


SynDrum, 2016-07-22:
Hi,
 I have the following problem(s) with my Radio Music Module. At first, when i pluged it in the first time, I just got the four top LED's flashing, maybe every half second, when i pulled out the SD Card, I had the of you discribed "4 lights flash together, and after a few more, you'll hear a thump at the module output as it goes into a slow cycle of auto reboots.". When i was wondering, whats wrong, I installed the hardwear-Check via Arduino, everything turned out to be fine, works as it should. Then, I wanted to reinstall the Radio_Musix.hex firmwear, which seems to work but now,  I neither have any sound-result, or the 4 lights flashing together, as it goes into the cycle of auto reboots. Any suggestions, what I should do know? The only thing thats on, is the Reset LED, and i have a little "one shot" sound when i pull out the SD Card,

Cheers, and thanks in Advance,

Niclas Helbig

TomWhitwell, 2016-07-23:
Hi, sorry to hear that. Things to check: 
What is on the SD card? Ensure it is known good, as here: https://github.com/TomWhitwell/RadioMusic/wiki/SD-Card:-Format-&-File-Structure

Is it a reputable brand SD Card 

Check soldering on the tiny pins on the SD Card holder

SynDrum, 2016-07-23:
Hi,  
Yeah, I'll try that. 
The brand should not be a prpblem, its a 16gb SanDisk, but i only formated it with my Smartephone, since I don't have a Micro-SD reader
Do you think the issue, with the firmwear thats not loading correctly anymore, has something to do with the circumstance, that i didnt cut the power supply? I read its "kind of" nessesary if you troubleshoot. Does it cause any serious damage to the teensy if i havent done so yet? :)

TomWhitwell, 2016-07-23:
You definitely need to load the SD Card properly. 

Re: cutting the trace & troubleshooting, it's quite simple, doesn't matter which you choose: 

If you CUT the trace, you need to supply 12v power from the ribbon cable AND USB power - this means you can run the module properly while it is connected to the computer. 

If you DO NOT CUT the trace, you an use 12v ribbon power OR USB power. So you can reprogram the Teensy just by connecting with USB cable, but you can't troubleshoot with the module active in your case. 

Either one works fine long term in the case.

TomWhitwell, 2016-12-12:
Closed because no response

closed 2016-12-12 by TomWhitwell