TomWhitwell/RadioMusic #104: Start point not at beginning + Background noise every time sample plays

starmandeluxe, 2016-09-08:
I built my module from Thonk and it works for the most part: the banks change, the samples play, reset and station CV works. But I've got a couple of nagging issues.

First is that if I change the station, the sample no longer starts at the beginning, but somewhere in the middle, so I have to turn the Start knob back to the bottom to get it to play from the beginning again. This is in loop: off mode.

Second is that every time the files actually play (reset or start is hit), there is a buzzing noise/hum in the background. I tried it on two power supplies (uZeus and Synthrotek noise filtering bus) and the issue happens on both. I also ensured with my meter that gnd is not continuous with +/- 12 and +12 is not continuous with -12. 

Any advice is really appreciated!

TomWhitwell, 2016-09-15:
First is that if I change the station, the sample no longer starts at the beginning, but somewhere in the middle, so I have to turn the Start knob back to the bottom to get it to play from the beginning again. This is in loop: off mode.
- this is normal behaviour https://github.com/TomWhitwell/RadioMusic/wiki/How-to-use-the-Radio-Music-module#station-knob

starmandeluxe, 2016-10-31:
Thanks for the response Tom! Hmm I'm not so sure that should be expected, according to the manual, the reset button or cv reset "restarts the current track at the point determined by the Start knob". Since I haven't moved the start knob  at all and always leave it at the lowest setting, it would seem strange that each time I change the station the sample resets to starting somewhere in the middle of the audio file. But I guess I can deal with it if it's not a mistake with my build.

TomWhitwell, 2016-12-12:
Closed because behaviour is normal

closed 2016-12-12 by TomWhitwell