TomWhitwell/RadioMusic #105: CV Reset Not Working

Chritian1968, 2016-09-11:
Hi 

I've built my Radio Music from Thonk and generally it's working fine but I cannot get any response from the CV input of the reset. I've tried everything from a clock to LFO, gate etc but it just doesn't respond. Have checked the soldering on everything it all looks good. Any ideas?

TomWhitwell, 2016-09-11:
It's not an issue I've heard before - likely to be a soldering issue or a wrong component - could D7 or D8 protection diodes be the wrong way around?

Chritian1968, 2016-09-11:
Hi Tom

They all seem to be fine and round the correct way. But I've tested step by step with a multi meter and it looks like the 220R R18 might be bead. I'll replace that and let you know.

Chritian1968, 2016-09-17:
Hi All

I've been trying to get to the bottom of why the CV reset won't work. I've replaced the 100k R19 & the 10K R8 and I'm getting good readings on all resistors. I've gone as far as bridging between the bottom leg of the CV jack input to the reset button side of the R8 and the reset cv suddenly works perfectly. And so the problem lies somewhere between the jack input and R8 but I can't se any obvious physical defect. Any clues, here's a photo. 
[image]

finnglink, 2022-10-24:
Hey @OceanWireAudio , did you ever get this to work?

OceanWireAudio, 2022-10-24:
Hi

Oh yes, I reflowed the resistor near the reset if I recall, the old rule “it’s always your soldering”

Regards

Christian

Sent from Christian's iPhone.