TomWhitwell/RadioMusic #106: pulsing buzzy sound plays through other modules even when not patched

jbook808, 2016-11-10:
Hi Tom, I recently assembled a DIY radio music kit and everything is great except there seems to be cross talk / interference traveling from the RM to other modules in my system. 

I've exhausted all troubleshooting skills I currently have and would greatly appreciate your help tracking down the issue.

Here is what I know:
• the sound is buzzy, almost like a pulsing square wave (maybe  clock signal coming from RM?)

• the sound loops with audio playing from the SD card. 
     • The sound does not play if audio is not playing, ie. with RM in one shot mode, pulse goes away when audio file finishes playback
     • removing SD card removes issue

• the sound glitches out and seems to restart rapidly as Channel Pot is moved

• reset button resets pulse loop

• the sound travels from RM to headphone out of Beface Output II module, but not main output. 

• the sound can be heard in the system when RM output is left unpatched
     • sound disappears if RM is removed from PSU

• the sound can be heard in the output section of Befaco BF22 filter when plugged into the same system.

• If the RM is loaded with test .hex files, the issue goes away. The issue is only present in RadioMusic1_1.cpp.hex, and RM_crossfade_2.0b3.hex. 
 
• I am using a Tip Top Audio uZues PSU
     • I've tried RM with a pittsburgh PSU and issue persisted

• I tested and found no continuity between +12v and GND... -12v and GND... or +12v and -12v

• I feel like the issue is grounding related. If I plug a stereo headphone jack into the RM out, I hear the pulse, which makes me think the pulse must be coming through the sleeve of the stereo headphone jack?

I took some pictures of the PCBs to show soldering. 

[image]
[image]
[image]
[image]
[image]

Thanks in advance for your help! I plan to upload audio examples tonight when I get home from work.

Best,
j

TomWhitwell, 2016-11-10:
Hi, sorry to hear that - what PSU are you using?

jbook808, 2016-11-10:
Hi!

I’m using a tip top audio PSU. I also tested with a pittsburgh PSU with same results.

Best,
Josh

jbook808, 2016-11-10:
Sorry for the double email. Tip Top Audio uZeus.

TomWhitwell, 2016-11-11:
uZeus can be tricky for grounding issues because of the cable busboards - that said I've used my uZeus to demo Radio Music at events very successfully. I can't see anything obvious - looks like a nice tidy build. 
I'm not really sure what to suggest - it's not an issue that I've heard elsewhere - it does sound like a symptom of grounding issues in the system. If you connect more sockets to the Radio Music (i.e. into reset or the CV inputs) does that have any effect on the noise (the patch cables act as another path to ground)

jbook808, 2016-11-11:
Below I've uploaded a few audio examples. There is a lot of hiss, but that's from other modules. 

RM is in one shot mode in all examples to show interference is present only when audio is playing back.

RM Interference 1.mp3 - Mono output of system with no patches made. Interference is triggered by reset button 2 times and then the station knob is turned. Station knob triggers reset button and illuminates reset LED. 

RM Interference 2.mp3 - RM Output patched into output module L. The interference is greatly amplified and makes sample playback unusable. 

RM Interference 3.mp3 - Patch into station from filter out, station modulates extremely fast and squelches.

RM Interference 4.mp3 - trigger input  going in to 'STATION'. This example shows station cv creates squelches as well as triggers reset button. 

[RM Interference Audio Examples.zip](https://github.com/TomWhitwell/RadioMusic/files/586543/RM.Interference.Audio.Examples.zip)

bscooper, 2016-11-17:
I finished my radio today, and all functions works nice. But i have also some rhythm sounds like "tac tac tac tac". Always, unimportant if a sound plays or not. It is like laying down a cell phone near then radio. I know that thismis something with the grounding in the power supply and in the connection between different modules. but why is the radio the bad guy? Which electronical part in the radio could be the problem? perhaps there is a way to fix with a better high quality part?

jbook808, 2016-11-17:
not sure! were you able to download the audio examples i uploaded above? does this noise sound like the noise you are experiencing?

Tom, do you think something like the Synthrotek Noise Filtering Power Distrobution Board might help with this issue?

NickShaffner, 2017-06-16:
whats that extra resistor you've got wired on the back of the bottom board?

I'm having much the same issue and am trying to track it down.   Did you ever find a resolution for this?

Building the think 2.1 full kit, I've gone over the pcb a few times and only found one of the signal diodes reversed which I replaced with a 4148 (docs say it's ok to do so).    I beeped through the board and got the scope out.   Looks like a 'reasonable' audio signal on the teensy dac pin which mimics the led  u meter.   However by the time the signal reaches the header pin between the front and rear it looks just like noise.

Using a Pittsburg modular supply and no other modules (this is my first eurorack module :) )

Will sleep on it and get the scope out again tomorrow.   It does feel a lot like a weird grounding issue.

Will do some more scoping tomorrow, but this

jbook808, 2017-06-20:
The extra resistor is a fix I found somewhere in the forums for an issue where noise quality would start to degrade after about 15 minutes of use. I added it in hopes that it would fix the noise that leaks between modules. No luck, though.