TomWhitwell/RadioMusic #109: start issue with reset input

equilet, 2017-03-08:
I'm having an issue with a Thonk kit that I built.

Setup:  
- clock into the [reset] jack
- no particular bank selected
- manually change [start] with knob

Symptom:
- Essentially, the issue is that the [start] position is respected based on the knob positioning, and functions as expected until around 1minute passes.
- Then, it seems that a new [start] position is "chosen" by the module.

Regression:  
- changing the start position again will jog the module back into respecting the knob position for a short time, and the problem will exhibit again

equilet, 2017-03-08:
Let me know if you need a hardware revision

tuuksijii, 2017-03-08:
I actually have the same exact issue, would love to know if there's anything to fix this.

equilet, 2017-03-23:
@TomWhitwell  - have you seen this issue before?

TomWhitwell, 2017-03-23:
@equilet No, I haven't seen it myself. 
Brainstorming it, see if @jstenroos has any ideas. 
It feels like software vs hardware, but the minute delay is unusual. 
I wondered if it could be the unsigned long that holds the start point overflowing, but quick calculation suggests that would only happen after 27 hours - and I'm not sure it's an issue anyway.

equilet, 2017-03-23:
AFAIK, it's not always exactly 1 minute.  It can be anywhere from ~45 seconds to 2 minutes.  I'll try to do some tests to see whether or not there's a statistically relevant window of time.

tuuksijii, 2017-03-23:
One thing that came to my mind is that could it be related to the length of the sample? I tried to reproduce it with a pad sample of around 10 seconds, but it kept the reset spot quite nicely.

jstenroos, 2017-03-23:
What firmware version are you using and what is the StartCVDivider value in the settings.txt?

It might be simply about the noise of the pot. They are normal analog pots, and are everything but stable. Raising the StartCVDivider value might help. 

-Jouni

equilet, 2017-03-23:
`StartCVDivider=2`

jstenroos, 2017-03-24:
Try 8 or 16.

TomWhitwell, 2017-03-24:
Yes, I think this might be the cause of this issue 
If you have a long sample; 10 minutes long, that means it has 25 million samples
The pot has a raw resolution of 1024 points. It's reasonable that over a minute, a reading of (say) 255 might become 256. That change in reading would move the start point by a second, from 2:29 to 2:30. 
The 'CV Divider' value reduces the granularity of the start CV input: 2 = 512 points 4 = 256 points. 
This means that  a CV input wobbling by a point or two is less likely to make any change to the start point. 
However, it does mean there may be unlucky times when this would mean that your CV is right between two boundaries; with a divider set to 4, that would give maybe a 3 second jump in the start point. 
If you have a 10 second sample, these changes are all much smaller; one point on the pot reflects 0.01 seconds in the start position. 

If you set StartCVImmediate to 1, you'll be able to hear the CV/pot wobbles in real time - because any change in CV or pot will immediately cause the sample to jump to that point. This might give an idea of the severity of the problem, and if you need to use shorter samples.

jstenroos, 2017-03-24:
In addition there is the time hysteresis value, I think I ended up with a value of 7. So, in practice it requires that the change exceeds +-7 to trigger a start position change. The StartCVDivider then kicks in and looks if we really are switching to a new position.

-Jouni

TomWhitwell, 2017-03-24:
I don't know if it's helpful, but I did the smoothing differently in Chord Organ, allowing gradual changes like this - seemed to work very well. 

    // Apply hysteresis and filtering to prevent jittery quantization 
    // Thanks to Matthias Puech for this code 

    if ((chordRaw > chordRawOld + 16) || (chordRaw < chordRawOld - 16)){
        chordRawOld = chordRaw;    
    }
    else {
        chordRawOld += (chordRaw - chordRawOld) >>5; 
        chordRaw = chordRawOld;  
    }

jstenroos, 2017-03-24:
Hmm, that’s a cool idea. Have to try it :)

-J

jamescarruthers, 2017-03-24:
I need to send you my PollPot library at some point - it takes care of this.

+44 (0)7834 828816

>
> I don't know if it's helpful, but I did the smoothing differently in
Chord Organ, allowing gradual changes like this - seemed to work very well.
>
> // Apply hysteresis and filtering to prevent jittery quantization
> // Thanks to Matthias Puech for this code
>
> if ((chordRaw > chordRawOld + 16) || (chordRaw < chordRawOld - 16)){
> chordRawOld = chordRaw;
> }
> else {
> chordRawOld += (chordRaw - chordRawOld) >>5;
> chordRaw = chordRawOld;
> }
https://github.com/TomWhitwell/RadioMusic/issues/109#issuecomment-288976043>,
or mute the thread <
https://github.com/

tuuksijii, 2017-04-04:
@jstenroos If you need beta-testing, I'd be very interested in helping out!

equilet, 2017-11-25:
I tested this and all is fine.  Thank you for your patience.

closed 2017-11-25 by equilet