TomWhitwell/RadioMusic #110: worked for a while...

lenick01, 2017-05-30:
I just build my Thonk kit Radio Music and it worked well immediately after the build.  However, it all of a sudden stopped working and three of the four top LEDs will blink intermittently and the lower LED stays on constantly - any ideas what happened?

Thonk-Steve, 2017-05-30:
It sounds like a classic case of cold solder joints to me, but can you provide more detail?

lenick01, 2017-05-30:
Sure;  I uploaded the sample files and initially the module did exactly
what I described before.  I cycled the power on and off a couple of times
and then it began working as expected for a couple of hours.  I had turned
the main power off again (Doepfer A100) and on and now it's stuck slowly
blinking three of the four top LEDs or none at all but the other LED stays
on constantly and it doesn't function.

Thonk-Steve, 2017-05-31:
So no audio? none of the CV inputs change anything?