TomWhitwell/RadioMusic #112: Sounds distorted. No pitch over samples. Sample doesn't play without reset trigger

Normalised, 2017-06-08:
From this thread : https://www.muffwiggler.com/forum/viewtopic.php?p=2573666#2573666

so i trued a couple of codes from polyfather without success , 

i install the official release as well,without great luck as well, 
now sounds distorted and no pitch over samples, also i have to trigger the sample now from the reset in in order to be played, i remember that wasnt the case when i got that thing to work first time.

Normalised, 2017-06-08:
I'll need all the details to work this out.

1. Exactly which hex file you used
2. The contents of settings.txt
3. The layout of the SD card folders and the contents
4. The brand and size of the SD card

pflab, 2017-06-08:
1.Exactly which hex file you used:

[Radio_Music_2017.1.zip](https://github.com/TomWhitwell/RadioMusic/files/1060911/Radio_Music_2017.1.zip)

2.The contents of settings.txt
i do not know what you mean by this one, sorry, happy to upload sth if you explain, do you wont me to open the code and upload the txt?sorry i have no clue from digital devices  

3.The layout of the SD card folders and the contents
Layout of SD cards, in both modules , are files from 0-15 , inside every folder .wav files randomly named

4.The brand and size of the SD card
SanDisk Ultra 16 GB MicroSDHC UHS-I

Normalised, 2017-06-08:
The 'settings.txt' is a file that is on the SD card which controls how the module behaves.

If you haven't downloaded one from the http://polyfather.com/radio_music/ application then there will be a default one on the SD card.
You can open this file in a text editor, e.g. notepad and copy and paste the contents in a comment here.

If you wish to try the 'pitch mode' then you need to turn on the option on the web page above, click 'download configuration' and copy the downloaded settings.txt to the SD card.

Thonk-Steve, 2017-06-08:
pflab on "sounds distorted"

maybe you knocked the trimmer on the back of the module by accident when upgrading software? try trimming that whilst playing sound to see if the distortion goes.

pflab, 2017-06-08:
ok

normalised, i will upload the settings,txt later today
i also think i was having the pitch mode on, but i was downloading and flashing the . hex files inly, never copied anything at the SD card, so it might be that.

steven , i ll have a look at the volume pot in case there is gain related from that

i will spend few hours after work 
thanks for the great support

pflab, 2017-06-08:
ok 
normalised. 
settings txt.

crossfade=0
crossfadeTime=25
showMeter=1
meterHide=2000
chanPotImmediate=1
chanCVImmediate=1
startPotImmediate=0
startCVImmediate=0
startCVDivider=2
looping=0
sort=1
pitchMode=0

pflab, 2017-06-08:
so 

i have news,

i made a new scenario at polyfather , 
downloaded the .hex but this time the settings.txt.
flashed the hex
copy pasted the NEW settings txt

and all is up and working now

Thonk-Steve, 2017-06-08:
Interesting, are you in Windows or OSX?

pflab, 2017-06-08:
though ,
is the sampler supposed to be played everytime a cv(here is pitch mode) is at start input, 
because at the moment I always have to trigger it from reset.
Or I have to make some choices at polyfather to choose that as an option, sorry if a have been annoying

thanks very much

pflab, 2017-06-08:
steven,

I am using Windows

Thonk-Steve, 2017-06-08:
How did you copy the new config over? Just drag it to the SD card in the
file explorer? As there any issue with the file not being writable or card
not being writable or did it just work?

pflab, 2017-06-08:
just delete the old settings.txt from the SD that was from the old script ;
and copy-paste the fresh ones 
and it just worked

Normalised, 2017-06-08:
For the 'change start cv to retrigger sample' you have to change this : 

startCVImmediate=0

to

startCVImmediate=1

Which on the polyfather app is the 'Start CV Immediate' option.

And no, you're not annoying. We're here to help as best we can :)

pflab, 2017-06-08:
lovely lovely,

one of my summer aims is to learn some coding 

all the best, and thanks to everyone that got involved on this project

petros

pflab, 2017-06-12:
guys still i can not get start iput to retrigger the sampler together with pitch
any ideas:

crossfade=1
crossfadeTime=50
showMeter=1
chanPotImmediate=1
chanCVImmediate=1
startPotImmediate=1
startCVImmediate=1
looping=0
anyAudioFiles=1
pitchMode=1
hardSwap=1
quantizeRootPot=0
quantizeRootCV=1
startCVDivider=1

Normalised, 2017-06-13:
But the button does retrigger?

What signal are you sending to the start input?

pflab, 2017-06-13:
Yes the button does retrigger, changing samples, I will check when I will be at home for sure. 

I have tried with few different signals digital s&h and sequencers to try to get v/Oct and retrigger

pflab, 2017-06-13:
yeap, i can verify that the button,
re-triggers and changing samples when pressed and input signal in reset,
when cv in start will change pitch , but nth is happening unleash there is a signal in reset

thanks

Thonk-Steve, 2017-06-13:
pflab could you record a quick video showing what you mean?

pflab, 2017-06-13:
good idea steve

https://youtu.be/TuSSReiOjX4

pflab, 2017-06-15:
/any news on this one, how does it looks to you ? ok ? i am start think i am doing sth wrong during the programming process

Normalised, 2017-06-16:
I'll look into it more this weekend.

pflab, 2017-06-17:
Normalised, you can have  a wee look but I am not gonna have Access to my synth for the next two weeks, so I won't be able to feedback with any more details for now. Thanks for all the help

Normalised, 2017-08-31:
First thing, I can see you have 2 RM modules, is it the same on both?

pflab, 2017-08-31:
yes, that is correct, both modules respond in the same way, one more sign i am doing sth wrong at the flashing script i guess

Normalised, 2017-09-10:
Ok, I think this all just a misunderstanding. I thought you were talking about the RESET input not retriggering the sample in pitch mode, but you're saying that you expect the START input to retrigger the sample in pitch mode. But in pitch mode the start input has no effect on start, its purely a pitch control AND changing the pitch DOES NOT retrigger the sample.

If you want to change it to also retrigger the sample you'll have to build the code yourself, the relevant parts are here :

https://github.com/TomWhitwell/RadioMusic/blob/069d035f33ae1ed93387e639efec3417c8bd05ba/RadioMusic/RadioMusic.ino#L320

closed 2017-09-10 by Normalised

pflab, 2017-09-10:
ok, and sorry for the misunderstanding 
thanks for all your patience, and support, 
thanks for a great project

Normalised, 2017-09-10:
No problem. Sorry it doesn't work like you expected, hopefully we'll be able to add this as an option in future, something like pitchCVRetrigger=1.