TomWhitwell/RadioMusic #115: Pitch on the start knob is always quantized

Normalised, 2017-06-18:
Pitch on the start knob is stepped/quantized even with quantizeRootPot=0

From : https://www.muffwiggler.com/forum/viewtopic.php?p=2578095#2578095

andrewpiano, 2017-06-19:
crossfade=1
crossfadeTime=185
showMeter=1
chanPotImmediate=1
chanCVImmediate=1
startPotImmediate=1
startCVImmediate=1
looping=1
anyAudioFiles=1
pitchMode=1
quantizeRootPot=0
quantizeRootCV=1
startCVDivider=0

andrewpiano, 2017-06-19:
I tried also "startCVDivider=1", not sure what that does but same results of start/root pot stepping.

andrewpiano, 2017-06-19:
Just wanted to say thanks for any help with this in advance. I hope it's something simple! I have a performance in a week where I'd really like to use the pitch feature with the Root Pot w/o quantization.

Normalised, 2017-06-19:
This should be fixed now with v2 : 

https://github.com/TomWhitwell/RadioMusic/releases/tag/v2017.2

It wasn't looking for the right names of the config options.

andrewpiano, 2017-06-19:
Yes~! Works wonderfully, thanks so much!

closed 2017-06-19 by Normalised

andrewpiano, 2017-06-19:
One last thing though, I notice that there is a sound descrepency between the Raw files and Wav files. Especially in non-quantized root pot mode, the wav files sound distorted and cannot seem to play in higher fidelity mode.

Normalised, 2017-06-20:
I noticed some artifacts when reaching the upper end of the pitch range but its more glitches rather than something i would call distortion.

if you have time could you upload a wav file that is a good example? also what would be really good is a recording of it playing the file so I can hear the type of distortion.

thanks.

andrewpiano, 2017-06-20:
Sure. Here is me adjusting the pitch of a solo piano file. Clouds on first half, then dry.
https://www.dropbox.com/s/2xicv9i4llukcu4/IMG-0392.mov?dl=0
This is the piano file original:
https://www.dropbox.com/s/8jfx6t2ep71hbf5/Radio%20Music%20piano2.1.wav?dl=0

Normalised, 2017-06-20:
great, thanks. i'll investigate.

andrewpiano, 2017-06-21:
I converted that piano file to Raw via audacity and it was even more distorted on RM playback. Hrm, the plot thickens... I'll try exporting the piano less hot next and test other wav files...

andrewpiano, 2017-06-21:
Other wav files like drum loops seem to play and pitch fine. So I guess it's something about the wav that I exported from Ableton?

andrewpiano, 2017-06-24:
Still trying to figure this out, can't seem to get a piano track out of Ableton to play nice in RM.