TomWhitwell/RadioMusic #119: Problems uploading firmware

Slomen, 2017-07-23:
The hex file with the blinking lights works as it should, but after uploading the actual radio music firmware I only get the "old" firmware even though I´ve tried different configurations in the config generator. I´ve tried it over and over, hope it is something stupid I´ve missed?

Thonk-Steve, 2017-07-23:
Hi, how are you identifying it as the old firmware still? What's not
working?

Slomen, 2017-07-23:
The start knob only change the start position and not the pitch of the sample. The functionality seems to be of the old firmware even though I've uploaded a firmware with the pitch mode.. 

Mvh Henrik Lindal

TomWhitwell, 2017-07-23:
Maybe download the new firmware again, from here: 
http://polyfather.com/radio_music/
And try once more

Normalised, 2017-07-23:
Also make sure you are copying the download configuration onto the SD card. The website doesn't configure the actual firmware itself but instead will let you download a separate text file that must be copied to the SD card.

Slomen, 2017-07-23:
That's the part I missed! Thank you so much!

Mvh Henrik Lindal

closed 2017-08-15 by TomWhitwell