TomWhitwell/RadioMusic #120: Possible to convert to run from 5v/9v?

23Gears, 2017-08-05:
I see that the schematic converts from a +12, -12v, Ground modular synth setup to a 5v power for the Teensy and associated circuits.

I am looking to build a mini portable modular setup. Is there a straightforward way to modify the module to run off 5v (or 9v)  DC?

TomWhitwell, 2017-08-05:
Hi, the teensy works fine on 5v but the audio out and CV in both require +\- 12v I'm afraid - it would be relatively straightforward to redesign the circuit, but you couldn't use the existing board

closed 2017-08-15 by TomWhitwell

23Gears, 2017-08-15:
Tom - thank you for replying to my post. Rude of me not to answer!

A friend and I are discussing a project to convert hte Radio Music to work
as a 9v unit which could be combined in a portable setup with such devices
as a Bastle Kastle. It will probably take me some time to get to the point
where I can work on this, but your answer was informative and helps me see
a possible way forward.

All the best - and thanks for such an intriguing machine!

Steve Cook

~
~~~~~~~~~~~~~~~~~~~~~~~~~~~~
steve@23gears.se
+46 (0)703-13 26 31
Twitter: @23gears <https://twitter.com/23Gears>
~~~~~~~~~~~~~~~~~~~~~~~~~~~~