TomWhitwell/RadioMusic #121: Scrolling through stations resets the sound files, is this normal?

mossgarden1, 2017-08-06:
Hi all, 

I've just built a RM, my first diy project and I'm amazed its up and running and working as far as I can tell. The only thing I've noticed is that whenever the station knob scrolls past a new station it resets to the begining. is this normal or is it something i can adjust with the firmware? How can I scroll through the stations without the files resetting? 

Thanks

TomWhitwell, 2017-08-06:
Hi, check here: https://github.com/TomWhitwell/RadioMusic/wiki/Customise-your-module%3A-Editing-settings.txt 
or, if you're using the latest firmware you can configure the settings.txt file and download it (and the firmware) here: 
http://polyfather.com/radio_music/

closed 2017-08-06 by TomWhitwell

eeweegh, 2017-08-06:
If you're using the latest firmware and have looping on, try adding loopMode=2 to your config file. Not sure that is a documented feature, but that's how the code seems to be written (loopMode==LOOP_MODE_CONTINUE, in AudioEngine.cpp, line 119)