TomWhitwell/RadioMusic #122: Radio Music Build Glitch Problem

10shun, 2017-08-24:
Hello I built a Radio Music kit and I seem to have a problem with it when I am triggering it.

What happens is I will be triggering a sample and then out of nowhere it will make this crazy glitch sound as if I am sending it a really fast audio rate trigger. It seems to happen randomly.

I am using the latest firmware from this site: https://github.com/TomWhitwell/RadioMusic/releases

and used the Configuration Generator.
I using the same firmware and configuration on 2 other radio music kits and they seem to function properly.

I have looked over the PCB to see if there are any bridges or missed solder spot and it seems alright.

Any help would be great.

10shun, 2017-08-24:
So I changed out the Micro sd card that was inside the module with the issue and put a different one inside of it and the issue seems to be gone.

I am thinking I have a bad micro sd card or I just have to reformat it.

I will report back to see if reformatting the sd card solves the problem

10shun, 2017-08-25:
So after reformatting the SD card I still get that glitch sound which pops up randomly and sounds like cv is changing the station or a really fast trigger.

I think this micro SD card is just messed up because when I insert a different one inside the module the problem is gone.

 I am using PNY 32GB Elite Micro SDHC.

I will report back with results when I purchase a new Card

TomWhitwell, 2017-08-26:
OK, that's why it's best to use quality branded cards: https://github.com/TomWhitwell/RadioMusic/wiki/SD-Card%3A-Format-%26-File-Structure

titotunes, 2017-11-25:
I'm getting this problem as well, and it happens when I'm sending a high voltage (around 5v) to the time/speed jack. I think some of that voltage is bleeding into the gate input.