TomWhitwell/RadioMusic #124: Wave Form button not working, Wave Form randomly changing when moving dials

kokodrimpy, 2017-08-24:
Hi,

I assembled the kit and it was working in perfect shape.
Until today :_(
By mistake I put a CV signal into the output, then I tried the module and the Wave Form button is not working, also the Wave Form randomly  s changing when moving the chord and root dials.

There is sound coming out of the module and it is coherent with the waveform displayed in the LEDs.

Please help!

Bests,
Pablo

kokodrimpy, 2017-08-24:
Here it is a video of the current module behaviour 

https://youtu.be/zbxVYSthG4U

actually it seems to be an issue with the trig input....

TomWhitwell, 2017-08-24:
Hi Pablo, 

Trig is an output - it pulses when chord changes, so you can trigger an envelope generator or whatever. 
Not sure what would happen if you put a signal into that - should be nothing, and it won't damage the module (in Radio Music, it's an input), but not sure what would happen. 
If you don't plug anything into trig, does it work normally? 

Tom

kokodrimpy, 2017-08-25:
Hi Tom,

Thanks a lot for your quick response. If nothing is connected to trig it behaves the same way.

There is nothing connected to trig at the beginning of the video https://www.youtube.com/watch?v=zbxVYSthG4U&feature=youtu.be 
Still, I get the wave forms changing randomly. 
I've just reformatted the teensy but I behaves the same.

Pablo

kokodrimpy, 2017-09-05:
Hi Tom,

Well, I tried everything I could think off. Formatted the SIM card, checked for cold soldering (didn't found any but still did some re-soldering in some points). I tested the button and it works properly. 

The only thing that i comes to my mind is either a controller or IC failure. Do you think this is more a TL074CN problem or an Teensy one?
Remember that all this seems to start happening after I put a CV signal into de audio output.
Is there any test that you would recommend to check for failures?

Please help!!! I was really loving my chord organ :_(

Bests,
Pablo.

oblivionratula, 2017-09-05:
That video is a trip because it's mirror-flipped. ;)  

I'm no expert, but the TL074 is socketed and cheap as chips, so I'd definitely try swapping that first. I'm inclined to think, looking at the schematics, if you killed the DAC with the voltage, you'd not be getting any sound now. On the other hand, if the OpAmp didn't like the voltage, I could see where messing up channel A or D might have strange effects on how channel B or C is feeding voltage to the Teensy. 

But again, I'm no expert.