TomWhitwell/RadioMusic #126: Whishlist: switch for LoopDevider Mode

piscocgn, 2017-09-28:
It would be great if there would be a switch in the default firmware to turn on a mode provided by the loop divider firmware. Reset becomes to output a clock signal which fits the loop.

piscocgn, 2018-08-03:
no rating or comment? more explanation required?

Thonk-Steve, 2018-08-03:
It's a nice idea, download the code and start hacking :)

piscocgn, 2018-08-03:
Your comment could imply that its possible in general and not eaten by hardware limitations to have it in place in general within the mainline firmware. With my 'basic' programming skills it would take ages. But let's see (Kids first, but maybe in a few years they are able to arise the competition ;-)

Thonk-Steve, 2018-08-03:
Yep, this is why you teach your kids to code ;)